nasa/fprime #5: Cornell Comments

timcanham, 2018-03-22:
•	Change "requirement.txt" to "required.txt" because that's what the actual file is named.
•	I had to manually install a few things not listed in the guide, but maybe the packages.sh file would have done that automatically.
•	Even though it tries to install lxml automatically, this hangs and doesn't appear to work. I had to do it manually and remove lxml from the required.txt to get it to proceed.
•	The demo is in "scripts" not "script".

harryharpel, 2019-02-16:
Hi, I'm a member of the Cornell CisLunar Explorers team. Is there an example repo of a CubeSat that implements fprime or any additional resources that JPL can provide to help members of the team learn more about fprime? Thanks!

ljreder, 2019-02-19:
There is a Ref example that comes with fprime.   There are not any open source cubesat examples that I know of.   This summer we will have a student potentially working on porting a balloon payload which is going to be come a Cubesat deployment.  If it happens I have agreement that this example will be open source but that will not be until the summer.  Perhaps email this question to fprime@jpl.nasa.gov for special consideration.

harryharpel, 2019-03-13:
When writing the xml file for a port is String an accepted type? The make make_gen and make commands are generating the appropriate files with no errors when I make the type of my argument an F32 but errors are generated when I make the type= "String". Any help would be greatly appreciated!

[image]

timcanham, 2019-03-13:
Use `string` with no caps, and a `size` attribute to define the length of the string. See `Autocoders/test/string_port` for an example.

harryharpel, 2019-03-13:
Thanks! That fixed the problem

LeStarch, 2020-03-03:
I am closing this issue, as I see no updates.  Please feel free to continue discussion on our mailing list:  https://groups.google.com/forum/#!forum/fprime-community

Or open a new issue if you detect a new bug.

closed 2020-03-03 by LeStarch