nasa/fprime #12: Binary Linking Error

ErynQian, 2018-10-16:
What might be the reason that "make ac_lvl1 ac_lvl2 ac_lvl3 ac_lvl4 " runs to completion but "make all" fails with a lot of "undefined reference to xxx" errors?
-----------------------------------------------
For more specific details:
1. We wrote a HelloWorld App which is very similar to MathComponent and its config in modules.mk is as follows:
`HELLOWORLD_MODULES := \
	HelloWorld/Top \
	HelloWorld/HelloTypes \
	HelloWorld/HelloPorts \
	HelloWorld/RecvBuffApp \
	HelloWorld/SendBuffApp \
	HelloWorld/SignalGen \
	HelloWorld/PingReceiver \
	HelloWorld/HelloSender \
	HelloWorld/HelloReceiver`
	
`HelloWorld_MODULES := \
	\
	$(HELLOWORLD_MODULES) \
	\
	$(SVC_MODULES) \
	\
	$(DEMO_DRV_MODULES) \
	\
	$(FW_MODULES) \
	\
	$(OS_MODULES) \
	\
	$(CFDP_MODULES) \
  	\
  	$(UTILS_MODULES)`

2. All libraries compiled successfully and all individual components passes the unit tests. But upon 'make all', the components seem to have trouble linking to the definitions of ports and types.
3. Error message:
[error.log](https://github.com/nasa/fprime/files/2481476/error.log)

timcanham, 2018-10-16:
Something funny is going on in the makefile:

```
make[1]: Entering directory '/home/eryn/fprime/HelloWorld'
/home/eryn/fprime/mk/makefiles/bin_targets.mk:12377: warning: overriding recipe for target '/home/eryn/fprime/HelloWorld/HelloTypes/linux-linux-x86-debug-gnu-bin/HelloSerializableAc.o'
/home/eryn/fprime/mk/makefiles/bin_targets.mk:12372: warning: ignoring old recipe for target '/home/eryn/fprime/HelloWorld/HelloTypes/linux-linux-x86-debug-gnu-bin/HelloSerializableAc.o'
/home/eryn/fprime/mk/makefiles/bin_targets.mk:12440: warning: overriding recipe for target '/home/eryn/fprime/HelloWorld/HelloPorts/linux-linux-x86-debug-gnu-bin/HelloPortAc.o'
/home/eryn/fprime/mk/makefiles/bin_targets.mk:12435: warning: ignoring old recipe for target '/home/eryn/fprime/HelloWorld/HelloPorts/linux-linux-x86-debug-gnu-bin/HelloPortAc.o'
/home/eryn/fprime/mk/makefiles/bin_targets.mk:12445: warning: overriding recipe for target '/home/eryn/fprime/HelloWorld/HelloPorts/linux-linux-x86-debug-gnu-bin/HelloBackPortAc.o'
/home/eryn/fprime/mk/makefiles/bin_targets.mk:12430: warning: ignoring old recipe for target '/home/eryn/fprime/HelloWorld/HelloPorts/linux-linux-x86-debug-gnu-bin/HelloBackPortAc.o'
Building LINUX
```

timcanham, 2018-10-16:
Can you post the `mod.mk` file for `/home/eryn/fprime/HelloWorld/HelloTypes`?

timcanham, 2018-10-16:
I think the undefined symbols are due to link order. The order in `HELLOWORLD_MODULES` is used as the link order. 

```
	HelloWorld/HelloTypes \
	HelloWorld/HelloPorts \
```
are listed before:
```
	HelloWorld/HelloSender \
	HelloWorld/HelloReceiver
```
Try reversing the order.

ErynQian, 2018-10-20:
That works.Thank you so much!

closed 2018-10-20 by ErynQian