nasa/fprime #17: MagicDraw18 SP6 on Windows velocity runtime failure

Shurtado2019, 2019-02-02:
Hello, 

We are running the fprime magicDraw plugin on MagicDraw18 SP6 Windows. We installed the JPL plugin and got the Component Autocoder to appear on toolbar. However when pressing Auto Generate Component/Port/Topology XML we get the following error: 

[image]

All of the XML files in AutoXML are generated but empty. Have you seen this issue before? What do you suggest we do next? 

Any help is greatly appreciated. Thank you!

LeStarch, 2019-02-03:
Is this on the "Ref" app, or a custom/modified app?  Have you tried to generate the files with this standard "Ref" app?

Also, looks like Windows.  Is this Win10?

Off the top of my head, that looks like a MagicDraw configuration issue.  Have you modified MagicDraw in any way?

LeStarch, 2019-02-04:
I'll take a look this evening, but we may need to setup a call or something.  One thing you could try right off the bat is to shorten the name.  "REFREFApplication.mdxml" may be too-long.  Try something like "ABCApplication.mdxml".  This is just a random hypothesis I had, but may be worth a try.

LeStarch, 2019-02-05:
Looks like I was able to duplicate the issue on my Win10 box here.  I am wondering if the plugin does not support windows machines at this stage because there are some bad paths of the form:

```
C:\User\mstarch\code\fprime/AutoXML/<some component>
```

Notice how the path separator changes part way through. I'll chat with the team tomorrow and see if I can get some information on this.

timcanham, 2019-02-05:
I use it pretty regularly on my Win10 box. There is something else going on...

LeStarch, 2019-02-05:
When you run MagicDraw is it 18.0 SP 6?  Or is it 18.5 SP 6?

timcanham, 2019-02-05:
It is 18.0 SP5.

LeStarch, 2019-02-05:
I meant for @Shurtado2019.  Are they using MagicDraw 18.0?

LeStarch, 2019-02-05:
@Shurtado2019 I think I have a solution for you.  Here is what I did:

```
cp -r <fprime>/Autocoders/MagicDrawCompPlugin/lib/ <magicdraw>/plugins/gov.nasa.jpl.magicdraw.componentAutocoder
```

Then edit your `plugin.xml` to include the following lines under "MyPlugin.jar" in the "<runtime>" tag.

```xml
           <library name="lib/commons-cli-1.3.1.jar"/>
           <library name="lib/velocity-1.6.2-dep.jar"/>
```

Full runtime tag:
```xml
        <runtime>
           <library name="MyPlugin.jar"/>
           <library name="lib/commons-cli-1.3.1.jar"/>
           <library name="lib/velocity-1.6.2-dep.jar"/>
        </runtime>
```

Please let me know if this fixes the problem.

LeStarch, 2019-02-05:
I tested this fix with 18.0 and 18.5 on our end. The old plugin and this fixed plugin produce the same results. However, with this fix, MD 18.5 is supported.

RaewynD, 2019-02-09:
> I meant for @Shurtado2019. Are they using MagicDraw 18.0?

We're using 18.0 SP 6 on Win 10, so unfortunately a nice mix of everyone's and not just one of these... Do we need to change our MagicDraw or did you end up getting the fix to work on 18.0 SP 6, too? If it did work on our setup, it'd be nice to call this week as it didn't work for us, sadly. Is there any day you're free after 6pm EST / 3pm PST?

Shurtado2019, 2019-02-09:
Also the About Message from the MagicDraw Component Plugin says it is Version 4.4, Plugin Generation Date: Fri Apr 21 20:03:04 EDT 2017.

LeStarch, 2019-02-09:
I'll e-mail y'all directly about setting up a call.

LeStarch, 2019-02-09:
I also pushed the latest plugin via PR. See: #18

timcanham, 2019-02-09:
Ergghhh. Didn’t realize it was a PR to master. We need to merge to devel.

Tim Canham
JPL Flight Software

LeStarch, 2019-02-10:
@timcanham sorry, I thought I specifically made a PR to devel.

timcanham, 2019-02-10:
Mistakenly merged to `master`. Merged from `master` to `devel`.

LeStarch, 2019-02-19:
Issue has been resolved.  The "<magic-draw-directory>/velocity.log" file was not writable.  Users of MagicDraw should, for the time being, ensure to have write permissions there, or run as an administrator.

Closing issue.

closed 2019-02-19 by LeStarch