nasa/fprime #21: make impl Command Not Recognized

brhs17, 2019-02-18:
My name is Benjamin and I am a new member on Cornell Cislunar Explorers.  I have been working thorough the math component tutorial, and at step 2.4.1 I cannot get the make impl command to create any stubs.

I run the command in the Ref/MathSender directory, and get the result:

make: *** No rule to make target 'impl'.  Stop.

I built the component without any problems, and have failed to find anything like misnamed files.

ljreder, 2019-02-19:
Perhaps try executing `make gen_make` and then retry the target.  I do not have it in front of me now but this might work ok.

timcanham, 2019-02-19:
@brhs17: Do you have the `Makefile` file in that directory? I'm looking at the tutorial and it doesn't explicitly tell you to put the file there. See section 2.1.1.3 at the description of the second make file that goes into each directory. This make file is not necessary for the overall builds, but provides some convenience targets local to the module directory so you don't have to keep jumping back and forth between the module directory and the directory where you build the deployment. If you create the make file in section 2.1.1.3 and make the variable `MODULE_DIR` equal to `Ref/MathSender`, the `make impl` target should work. We need to update the tutorial to add this step to create `Makefile` in each directory.

Here is what the `Makefile` should look like:

```
# derive module name from directory

MODULE_DIR = Ref/MathSender
MODULE = $(subst /,,$(MODULE_DIR))

BUILD_ROOT ?= $(subst /$(MODULE_DIR),,$(CURDIR))
export BUILD_ROOT

include $(BUILD_ROOT)/mk/makefiles/module_targets.mk
```

brhs17, 2019-02-22:
@timcanham The missing makefile was the problem.
Thank you.

julia861127, 2019-02-24:
I tried to add a makefile in my sender directory, however, when I tried `make impl` again, it shows the message: **Makefile:9: %(CURDIR)/mk/makefiles/module_targets.mk: No such file or directory**

LeStarch, 2019-02-24:
In your makefile, make sure the line: 
```
MODULE_DIR = ...
```
Is updated for the location of your module.  Also, typically, `%(CURDIR)` is `%(BUILD_ROOT)/` in our Makefiles.

If this doesn't help, can you post your full makefile here for us to help debug?

julia861127, 2019-02-25:
```
# derive module name from directory

MODULE_DIR = Ref/EPSPower/EPSPowerSender
MODULE = $(subst /,,$(MODULE_DIR))

BUILD_ROOT ?= $(subst /$(MODULE_DIR),,%(CURDIR))
export BUILD_ROOT

include $(BUILD_ROOT)/mk/makefiles/module_targets.mk
```

LeStarch, 2019-02-25:
This Makefile expects to see the directory `EPSPowerSender` at `fprime/Ref/EPSPower/EPSPowerSender`, where fprime is the base of your F´ install.  Make sure the above Makefile is in that directory.

julia861127, 2019-02-25:
Yes, the above makefile is in the `fprime/Ref/EPSPower/EPSPowerSender` directory, but the `make impl` was not working with the message shown above.

timcanham, 2019-02-25:
I think you need to change `%(CURDIR)` to `$(CURDIR)`.

wr3484, 2019-03-09:
I am on section 2.1.1.3 of the tutorial, and continue to receive this:
[image]

'''
# derive module name from directory

MODULE_DIR = Ref\MathPorts
MODULE = $(subst /,,$(MODULE_DIR))

BUILD_ROOT ?= $(subst /$(MODULE_DIR),,$(CURDIR))
export BUILD_ROOT

include $(BUILD_ROOT)/mk/makefiles/module_targets.mk
'''

timcanham, 2019-03-09:
Is the above `Makefile.mk`? If so, it should be `Makefile` so `make` automatically finds it. Alternatively, `make -f Makefile.mk gen_make`

wr3484, 2019-03-09:
Thank you for clarifying that, after I changed it though I am still not able to build.
[image]

This was the change made  in the modules.mk 
```
REF_MODULES := \
	Ref/Top \
	Ref/RecvBuffApp \
	Ref/SendBuffApp \
	Ref/SignalGen \
	Ref/PingReceiver \
	Ref/MathPorts
```

timcanham, 2019-03-09:
It can't find Cheetah. Based on the path, it looks like you may be trying the Cygwin build? Fix the path in `/mk/configs/host/CYGWIN-generic.mk`. Fix it to `cheetah-compile` with no path. If you installed it, it should be in your path.

wr3484, 2019-03-10:
Apologies,I don't follow where to change the path, is that it where I put a break point?
[image]

timcanham, 2019-03-10:
Yes, you found the correct file. Comment in line five.

timcanham, 2019-03-10:
If you are using cygwin, it would be in your cygwin path, not the Windows install. You need to `pip install` from the cygwin shell.

wr3484, 2019-03-10:
After running a pip install of cheetah from my CYGWIN terminal everything worked. **THANK YOU**.

wr3484, 2019-03-11:
Attempting to run the Demo from fPrime Manual and having build errors using make, using CYGWIN and have installed all needed packages.

[image]

[image]

rdaruwala, 2019-03-11:
@wr3484 It looks like your `make` command is attempting to build for `LINUX` instead of `CYGWIN`. Could you try running `make CYGWIN` and see if that helps?

timcanham, 2019-03-11:
It also looks like the version encoder doesn't like it if there are no commits yet. Is this just a copy of the source rather than a fork? At any rate, do at least one commit of the code. We should make it more robust to non-git copies, however.

wr3484, 2019-03-11:
The `make CYGWIN` command worked, now it is not recognizing Tkinter? I have python 2.7 and 3.7 installed. I thought python outdated Tkinter and switched to tkinter? 
[image]

timcanham, 2019-03-12:
One of our efforts that will be in a future update for F' will be advancing Python to 3.x, but for now it requires 2.x. I haven't really tried running with both yet.

timcanham, 2019-03-12:
Also, Cygwin isn't a supported platform. We support Linux and MacOS. We used to support Cygwin, but in the last year or so switched to WSL for Windows. If you want to do PRs with Cygwin fixes, we'll accept them, but they will be as-is.

brhs17, 2019-03-15:
I am trying to make my own component and am getting this error when I try to `make impl`
[image]

rdaruwala, 2019-03-15:
In your `CLEMOpsComponentAi.xml` file, line 46, `ACTIVITY_HIGH` should be `ACTIVITY_HI`

brhs17, 2019-03-15:
I fixed that error, but there are more.  Is there anyway to figure out where the recipe failed?

rdaruwala, 2019-03-15:
I was able to generate template files after your component after fixing that one error. Are you still having issues generating those files, or are you trying to make something else?

Regarding your second question, when I first tried running `make` on your code, I received a tad more descriptive error message:

```
XML file CLEMOpsComponentAi.xml is not valid according to schema fprime/Autocoders/schema/default/component_schema.rng.
Extra element events in interleave
<string>:0:0:ERROR:RELAXNGV:RELAXNG_ERR_INTEREXTRA: Extra element events in interleave
fprime/Ref/CLEMOps/CLEMOpsComponentAi.xml:1:0:ERROR:RELAXNGV:RELAXNG_ERR_CONTENTVALID: Element component failed to validate content
fprime/Ref/CLEMOps/CLEMOpsComponentAi.xml:1:0:ERROR:RELAXNGV:RELAXNG_ERR_CONTENTVALID: Element component failed to validate content
<lxml.etree._ElementTree object at 0x7fd26de94a70>
Traceback (most recent call last):
  File "fprime/Autocoders/bin/codegen.py", line 1125, in <module>
    main()
  File "fprime/Autocoders/bin/codegen.py", line 1086, in main
    the_parsed_component_xml = XmlComponentParser.XmlComponentParser(xml_filename)
  File "fprime/Autocoders/src/parsers/XmlComponentParser.py", line 104, in __init__
    raise e
AssertionError: Extra element events in interleave
```

I think you could get a similar output by appending `VERBOSE=1` to your make command

brhs17, 2019-03-15:
I still get the same error and error message after fixing `ACTIVITY_HI` and adding `VERBOSE=1`
[image]

timcanham, 2019-03-18:
Some basic questions: 

1) Did you add `CLEMOps` to `mk/configs/modules/modules.mk`?
2) Did you invoke `make gen_make` after you added `CLEMOps` to 1)?

brhs17, 2019-03-18:
Thankyou, I forgot to do the `make gen_make`.

brianthespoon, 2019-03-18:
Hello Tim,I was trying to set up a virtual environment where I would use fprime, but it seems fprime is pointing to my system-level python dependencies looking for lxml (which fails on run of make because i do t have lxml at the system level). I would rather point it to the virtual environment. Can you point me to the file where I can modify this? Sent from my Verizon, Samsung Galaxy smartphone

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/nasa/fprime/issues/21#issuecomment-473964251",
"url": "https://github.com/nasa/fprime/issues/21#issuecomment-473964251",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

brhs17, 2019-03-22:
How do I find errors in my XML files?  My debug response tells me which file is wrong, but nothing about where it is wrong, always saying something like `extra element events in interleave`

closed 2019-03-28 by brhs17