nasa/fprime #22: make ut - No rule to make target

brhs17, 2019-02-25:
[image]
I've checked all of the makefiles and as far as I can tell they are identical to the tutorial code.

timcanham, 2019-02-25:
This looks like something different - it looks like `Ref/MathPort` might be missing from `mk/configs/modules/modules.mk`.

brhs17, 2019-02-25:
This is what I have, looks right.  The directory is name MathPorts.
[image]

timcanham, 2019-02-25:
Your `module.mk` has `MathPorts` (plural), while your `MathSender/test/ut/mod.mk` looks like it has `MathPort` (singular). I'm guessing to have to fix the `mod.mk` in `test/ut`.

brhs17, 2019-02-27:
That wasn't the problem, my mod.mk file had it plural as well.  I see the singular call you are talking about in the error but I have no idea where it could come from.
[image]

Brian-Campuzano, 2019-02-27:
Try running the following command to try to figure out where the `MathPort`  target is coming from:

```
grep -r MathPort | grep -v MathPorts
```

brhs17, 2019-02-27:
Seems to be some document in the makefile folder that I shouldn't mess with.  Do I need to rebuild or something?
[image]

Brian-Campuzano, 2019-02-27:
Yes I would execute a `make rebuild` in the fprime/ref directory.

In general its a good idea to rebuild the build system after changing mod.mk files.

brhs17, 2019-02-27:
Ok, when I was trying to debug earlier I remember finding a singular MathPort somewhere and fixing it, but  I didn't realize I needed to rebuilt before it would do anything, so I ran "make rebuild" in the ref directory.  I'm getting a different error now: 
[image]

Brian-Campuzano, 2019-02-27:
Looks like compilation of `Ref/MathSender/test/ut/main.c` failed but the error message is cutoff.

brhs17, 2019-02-28:
I fixed the compiling errors, but now I seem to get a permission error when I try to `run_ut`
[image]

timcanham, 2019-02-28:
You need to do a `chmod u+x runtest_LINUX` to set the executable bit to run the shell script.

brhs17, 2019-03-01:
I'm trying to run `make ut` in Math Receiver, where I've copied the tutorial files into ut, but when I try to build it the program claims that `m_factor1` does not exist.
[image]

brhs17, 2019-03-01:
I type `m_factor1` wrong in the set factor 1 cmd handler, but now that I've fixed it and tried to rebuild it still returns the same error.

brhs17, 2019-03-01:
I looked through the tutorial and couldn't find a mention of m_factor1 I didn't have and the tutorial version of `MathReceieverComponentImpl.cpp` results in the same errors, and another one.

timcanham, 2019-03-01:
I see it here:

https://github.com/nasa/fprime/blob/master/docs/Tutorials/MathComponent/MathReceiver/MathReceiverComponentImpl.hpp#L102

It looks like we need to add header file descriptions to the demo.

brhs17, 2019-03-01:
Adding the indicated variable allows the program to build successfully.  However, running the test cases results in some extremely strange values:
[image]
All four of the math tests failed on the same assert statement in each case.
The offending line in tester.cpp for the math cases was
`ASSERT_TLM_MR_FACTOR1S(0,1);`
and for the throttle test:
`ASSERT_TLM_MR_FACTOR1S(0,cycle+1);`

brhs17, 2019-03-01:
The large numbers change between multiple runs.

timcanham, 2019-03-02:
The other thing: Did you initialize the values in the constructor?

brhs17, 2019-03-02:
When I try to add that it doesn't compile:
[image]

timcanham, 2019-03-02:
Can you post the code fragment? It should work.

brhs17, 2019-03-02:
[image]

timcanham, 2019-03-02:
`,m_factor(0.0)`, not `,m_factor(0,0)` :)

brhs17, 2019-03-02:
4/5 Test cases passed, but there seems to be a problem with `paramSend_factor2`
[image]

timcanham, 2019-03-04:
I just re-ran the unit tests based on the code checked in to `docs/Tutorials/MathComponent`. Did you diff against that code?

brhs17, 2019-03-05:
Ok, I didn't put in the parameter updated function, but when I put it in I get this error.
[image]
I don't know how to "override a virtual function" as listed in the tutorial.

brhs17, 2019-03-05:
There is this line in MathReceiverComponent.hpp that looks like it should be declaring the function:
[image]

timcanham, 2019-03-06:
Can you post the function definition in the .cpp file? You should also take off the `virtual` on the declaration in `MathReceiverComponent.hpp`. This error usually means something in the definition doesn't match the declaration in the header file.

brhs17, 2019-03-06:
[image]

timcanham, 2019-03-06:
This should work - is this in a fork somewhere? The only thing I can think of is that the declaration is outside the class definition or something.

timcanham, 2019-03-06:
Also, a GitHub tip: you can post code with triple tics "```" + "c++" and get nicely formatted code:

```c++
namespace Svc {

    //! \class CommandDispatcherImpl
    //! \brief Command Dispatcher component class
    //!
    //! The command dispatcher takes incoming Fw::Com packets that contain
    //! encoded commands. It extracts the opcode and looks it up in a table
    //! that is populated by components at registration time. If a component
    //! is connected to the seqCmdStatus port with the same number
    //! as the port that submitted the command, the command status will be returned.

    class CommandDispatcherImpl : public CommandDispatcherComponentBase {
        public:
            //!  \brief Command Dispatcher constructor
            //!
            //!  The constructor initializes the state of the component.
            //!  In this component, the opcode dispatch and tracking tables
            //!  are initialized.
            //!
            //!  \param name the component instance name
#if FW_OBJECT_NAMES == 1
            CommandDispatcherImpl(const char* name);
#else
            CommandDispatcherImpl();
#endif
            //!  \brief Component initialization routine
            //!
            //!  The initialization function calls the initialization
            //!  routine for the base class.
            //!
            //!  \param queueDepth the depth of the message queue for the component
            void init(
                    NATIVE_INT_TYPE queueDepth, /*!< The queue depth*/
                    NATIVE_INT_TYPE instance /*!< The instance number*/
                    ); //!< initialization function
            //!  \brief Component destructor
            //!
            //!  The destructor for this component is empty
            virtual ~CommandDispatcherImpl();
```

brhs17, 2019-03-06:
I don't have a fork, I'm not sure where I would put it.  I'm pretty sure it is in the class definition since the ending bracket is after it.  How is the best way to let you look at the files?

brhs17, 2019-03-06:
I uploaded my code onto a branch at this link I think: https://github.com/brhs17/fprime/tree/MathTutorial/Ref

timcanham, 2019-03-06:
I looked at your branch, and I don't see the function definition in `MathReceiverComponent.hpp`.

brhs17, 2019-03-06:
Ok, got it to work.  What does the Ac vs Impl files do differently, I'm not sure when I have to define things in which file, because the definitions of `parameterUpdated` are very similar between the files.  Why do I need them both?

timcanham, 2019-03-06:
In C++ when you define a virtual function in a base class, the derived class has to redeclare it in order to override it. It's a C++ thing. In F`, we make the handlers for ports and commands a pure virtual, which requires that the derived class override it. It's a way to make sure the developer doesn't forget to implement a handler.

brhs17, 2019-03-07:
I went through section 3 of the tutorial and the `make` command in `Ref/Top` seems successful, but I get an error trying to build Ref total
[image]

timcanham, 2019-03-07:
Have you uploaded the changes to `/mk/configs/modules/modules.mk`? It may be a linking order issue there.

brhs17, 2019-03-07:
My `MathTutorial` branch should be up to date.  I didn't see any changes that needed to be made to `modules.mk` in this section of the tutorial?

timcanham, 2019-03-07:
I see the entries in your latest push. I think it is an ordering issue. You have:

```
REF_MODULES := \
	Ref/Top \
	Ref/RecvBuffApp \
	Ref/SendBuffApp \
	Ref/SignalGen \
	Ref/PingReceiver \
	Ref/MathPorts \
	Ref/MathTypes \
	Ref/MathSender \
	Ref/MathReceiver
```

It should be:

```
REF_MODULES := \
	Ref/Top \
	Ref/RecvBuffApp \
	Ref/SendBuffApp \
	Ref/SignalGen \
	Ref/PingReceiver \
	Ref/MathSender \
	Ref/MathReceiver \
	Ref/MathPorts \
	Ref/MathTypes
```

Specifically, the `Ref/MathPorts` and `Ref/MathTypes` have to be after the `Ref/MathSender` and `Ref/MathReceiver` in the link order. This is not an F` thing, it is a linker thing.

brhs17, 2019-03-07:
Ok, I got the math tutorial mostly working.  All the math operations and factors work correctly, but the `Log Events` and 'Channel Telemetry' tabs seem to stop working once I try to call the `doMath` command.  The commands still operate correctly in the extra window, just the gio stops working correctly.
[image]

timcanham, 2019-03-07:
Let me look into the error. This rings a bell.

rdaruwala, 2019-03-11:
@brhs17 This can be fixed by changing the following line:

`<event id="0" name="MS_COMMAND_RECV" severity="ACTIVITY_LO" format_string="Math Cmd Recvd: %f %d %f"  >`

to

`<event id="0" name="MS_COMMAND_RECV" severity="ACTIVITY_LO" format_string="Math Cmd Recvd: %f %f %d"  >`

in `Ref/MathSender/MathSenderComponentAi.xml`

closed 2019-03-28 by brhs17