nasa/fprime #23: Problem when make at the end in the Ref folder

vigno88, 2019-02-27:
HI,

I have this error when I tried to `make` at the end of the math tutorial in the `Ref` folder. Every module are built with no errors but when it is time to `make` the `Ref` folder, I get this mistake. I tried to make my own components and I got the same errors. Could you please guide me? 

I copied exactly every files from the Tutorial folder so there shouldn't be any problem.
```shell
Linking binary...
/home/nathan/dev/fprime_math/fprime/Ref/MathSender/linux-linux-x86-debug-gnu-bin/libRefMathSender.a(MathSenderComponentAc.o): In function `Ref::MathSenderComponentBase::set_mathOut_OutputPort(int, Ref::InputMathOpPort*)':
/home/nathan/dev/fprime_math/fprime/Ref/MathSender/MathSenderComponentAc.cpp:116: undefined reference to `Ref::OutputMathOpPort::addCallPort(Ref::InputMathOpPort*)'
/home/nathan/dev/fprime_math/fprime/Ref/MathSender/linux-linux-x86-debug-gnu-bin/libRefMathSender.a(MathSenderComponentAc.o): In function `Ref::MathSenderComponentBase::MathSenderComponentBase(char const*)':
/home/nathan/dev/fprime_math/fprime/Ref/MathSender/MathSenderComponentAc.cpp:281: undefined reference to `Ref::InputMathResultPort::InputMathResultPort()'
/home/nathan/dev/fprime_math/fprime/Ref/MathSender/MathSenderComponentAc.cpp:281: undefined reference to `Ref::OutputMathOpPort::OutputMathOpPort()'
/home/nathan/dev/fprime_math/fprime/Ref/MathSender/linux-linux-x86-debug-gnu-bin/libRefMathSender.a(MathSenderComponentAc.o): In function `Ref::MathSenderComponentBase::init(int, int)':
/home/nathan/dev/fprime_math/fprime/Ref/MathSender/MathSenderComponentAc.cpp:313: undefined reference to `Ref::InputMathResultPort::addCallComp(Fw::PassiveComponentBase*, void (*)(Fw::PassiveComponentBase*, int, float))'
/home/nathan/dev/fprime_math/fprime/Ref/MathSender/linux-linux-x86-debug-gnu-bin/libRefMathSender.a(MathSenderComponentAc.o): In function `Ref::MathSenderComponentBase::mathOut_out(int, float, float, Ref::MathOperation)':
/home/nathan/dev/fprime_math/fprime/Ref/MathSender/MathSenderComponentAc.cpp:542: undefined reference to `Ref::OutputMathOpPort::invoke(float, float, Ref::MathOperation)'
/home/nathan/dev/fprime_math/fprime/Ref/MathReceiver/linux-linux-x86-debug-gnu-bin/libRefMathReceiver.a(MathReceiverComponentImpl.o): In function `Ref::MathReceiverComponentImpl::mathIn_handler(int, float, float, Ref::MathOperation)':
/home/nathan/dev/fprime_math/fprime/Ref/MathReceiver/MathReceiverComponentImpl.cpp:73: undefined reference to `Ref::MathOp::MathOp()'
/home/nathan/dev/fprime_math/fprime/Ref/MathReceiver/MathReceiverComponentImpl.cpp:77: undefined reference to `Ref::MathOp::setop(Ref::Operation)'
/home/nathan/dev/fprime_math/fprime/Ref/MathReceiver/MathReceiverComponentImpl.cpp:81: undefined reference to `Ref::MathOp::setop(Ref::Operation)'
/home/nathan/dev/fprime_math/fprime/Ref/MathReceiver/MathReceiverComponentImpl.cpp:85: undefined reference to `Ref::MathOp::setop(Ref::Operation)'
/home/nathan/dev/fprime_math/fprime/Ref/MathReceiver/MathReceiverComponentImpl.cpp:89: undefined reference to `Ref::MathOp::setop(Ref::Operation)'
/home/nathan/dev/fprime_math/fprime/Ref/MathReceiver/MathReceiverComponentImpl.cpp:99: undefined reference to `Ref::MathOp::setval1(float)'
/home/nathan/dev/fprime_math/fprime/Ref/MathReceiver/MathReceiverComponentImpl.cpp:100: undefined reference to `Ref::MathOp::setval2(float)'
/home/nathan/dev/fprime_math/fprime/Ref/MathReceiver/MathReceiverComponentImpl.cpp:101: undefined reference to `Ref::MathOp::setresult(float)'
/home/nathan/dev/fprime_math/fprime/Ref/MathReceiver/MathReceiverComponentImpl.cpp:102: undefined reference to `Ref::MathOp::MathOp(Ref::MathOp const&)'
/home/nathan/dev/fprime_math/fprime/Ref/MathReceiver/linux-linux-x86-debug-gnu-bin/libRefMathReceiver.a(MathReceiverComponentImpl.o): In function `Ref::MathOp::~MathOp()':
/home/nathan/dev/fprime_math/fprime/Ref/MathTypes/MathOpSerializableAc.hpp:32: undefined reference to `vtable for Ref::MathOp'
/home/nathan/dev/fprime_math/fprime/Ref/MathReceiver/linux-linux-x86-debug-gnu-bin/libRefMathReceiver.a(MathReceiverComponentAc.o): In function `Ref::MathReceiverComponentBase::set_mathOut_OutputPort(int, Ref::InputMathResultPort*)':
/home/nathan/dev/fprime_math/fprime/Ref/MathReceiver/MathReceiverComponentAc.cpp:124: undefined reference to `Ref::OutputMathResultPort::addCallPort(Ref::InputMathResultPort*)'
/home/nathan/dev/fprime_math/fprime/Ref/MathReceiver/linux-linux-x86-debug-gnu-bin/libRefMathReceiver.a(MathReceiverComponentAc.o): In function `Ref::MathReceiverComponentBase::MathReceiverComponentBase(char const*)':
/home/nathan/dev/fprime_math/fprime/Ref/MathReceiver/MathReceiverComponentAc.cpp:382: undefined reference to `Ref::InputMathOpPort::InputMathOpPort()'
/home/nathan/dev/fprime_math/fprime/Ref/MathReceiver/MathReceiverComponentAc.cpp:382: undefined reference to `Ref::OutputMathResultPort::OutputMathResultPort()'
/home/nathan/dev/fprime_math/fprime/Ref/MathReceiver/linux-linux-x86-debug-gnu-bin/libRefMathReceiver.a(MathReceiverComponentAc.o): In function `Ref::MathReceiverComponentBase::init(int, int)':
/home/nathan/dev/fprime_math/fprime/Ref/MathReceiver/MathReceiverComponentAc.cpp:417: undefined reference to `Ref::InputMathOpPort::addCallComp(Fw::PassiveComponentBase*, void (*)(Fw::PassiveComponentBase*, int, float, float, Ref::MathOperation))'
/home/nathan/dev/fprime_math/fprime/Ref/MathReceiver/linux-linux-x86-debug-gnu-bin/libRefMathReceiver.a(MathReceiverComponentAc.o): In function `Ref::MathReceiverComponentBase::mathOut_out(int, float)':
/home/nathan/dev/fprime_math/fprime/Ref/MathReceiver/MathReceiverComponentAc.cpp:718: undefined reference to `Ref::OutputMathResultPort::invoke(float)'
/home/nathan/dev/fprime_math/fprime/Ref/MathReceiver/linux-linux-x86-debug-gnu-bin/libRefMathReceiver.a(MathReceiverComponentAc.o): In function `Ref::MathReceiverComponentBase::log_ACTIVITY_HI_MR_OPERATION_PERFORMED(Ref::MathOp)':
/home/nathan/dev/fprime_math/fprime/Ref/MathReceiver/MathReceiverComponentAc.cpp:1676: undefined reference to `Ref::MathOp::toString(Fw::StringBase&) const'
collect2: error: ld returned 1 exit status
/home/nathan/dev/fprime_math/fprime/mk/makefiles/Makefile:63: recipe for target '/home/nathan/dev/fprime_math/fprime/Ref/linux-linux-x86-debug-gnu-bin/Ref' failed
make[1]: *** [/home/nathan/dev/fprime_math/fprime/Ref/linux-linux-x86-debug-gnu-bin/Ref] Error 1
make[1]: Leaving directory '/home/nathan/dev/fprime_math/fprime/Ref'
Command exited with non-zero status 2
Build Time: 0:01.84
/home/nathan/dev/fprime_math/fprime/mk/makefiles/deployment_makefile.mk:48: recipe for target 'LINUX' failed
make: *** [LINUX] Error 2

```

timcanham, 2019-02-27:
Check the order in `mk/configs/modules/modules.mk`. The `MathPorts` directory should be below `MathSender` and `MathReceiver`. If we made that mistake in the tutorial, let us know.

vigno88, 2019-02-27:
In this order:
```Makefile
REF_MODULES := \
	Ref/Top \
	Ref/RecvBuffApp \
	Ref/SendBuffApp \
	Ref/SignalGen \
	Ref/PingReceiver \
	Ref/MathTypes \
	Ref/MathSender \
	Ref/MathReceiver \
	Ref/MathPorts 
```
I get this : 
```shell
Linking binary...
/home/nathan/dev/fprime_math/fprime/Ref/MathReceiver/linux-linux-x86-debug-gnu-bin/libRefMathReceiver.a(MathReceiverComponentImpl.o): In function `Ref::MathReceiverComponentImpl::mathIn_handler(int, float, float, Ref::MathOperation)':
/home/nathan/dev/fprime_math/fprime/Ref/MathReceiver/MathReceiverComponentImpl.cpp:73: undefined reference to `Ref::MathOp::MathOp()'
/home/nathan/dev/fprime_math/fprime/Ref/MathReceiver/MathReceiverComponentImpl.cpp:77: undefined reference to `Ref::MathOp::setop(Ref::Operation)'
/home/nathan/dev/fprime_math/fprime/Ref/MathReceiver/MathReceiverComponentImpl.cpp:81: undefined reference to `Ref::MathOp::setop(Ref::Operation)'
/home/nathan/dev/fprime_math/fprime/Ref/MathReceiver/MathReceiverComponentImpl.cpp:85: undefined reference to `Ref::MathOp::setop(Ref::Operation)'
/home/nathan/dev/fprime_math/fprime/Ref/MathReceiver/MathReceiverComponentImpl.cpp:89: undefined reference to `Ref::MathOp::setop(Ref::Operation)'
/home/nathan/dev/fprime_math/fprime/Ref/MathReceiver/MathReceiverComponentImpl.cpp:99: undefined reference to `Ref::MathOp::setval1(float)'
/home/nathan/dev/fprime_math/fprime/Ref/MathReceiver/MathReceiverComponentImpl.cpp:100: undefined reference to `Ref::MathOp::setval2(float)'
/home/nathan/dev/fprime_math/fprime/Ref/MathReceiver/MathReceiverComponentImpl.cpp:101: undefined reference to `Ref::MathOp::setresult(float)'
/home/nathan/dev/fprime_math/fprime/Ref/MathReceiver/MathReceiverComponentImpl.cpp:102: undefined reference to `Ref::MathOp::MathOp(Ref::MathOp const&)'
/home/nathan/dev/fprime_math/fprime/Ref/MathReceiver/linux-linux-x86-debug-gnu-bin/libRefMathReceiver.a(MathReceiverComponentImpl.o): In function `Ref::MathOp::~MathOp()':
/home/nathan/dev/fprime_math/fprime/Ref/MathTypes/MathOpSerializableAc.hpp:32: undefined reference to `vtable for Ref::MathOp'
/home/nathan/dev/fprime_math/fprime/Ref/MathReceiver/linux-linux-x86-debug-gnu-bin/libRefMathReceiver.a(MathReceiverComponentAc.o): In function `Ref::MathReceiverComponentBase::log_ACTIVITY_HI_MR_OPERATION_PERFORMED(Ref::MathOp)':
/home/nathan/dev/fprime_math/fprime/Ref/MathReceiver/MathReceiverComponentAc.cpp:1676: undefined reference to `Ref::MathOp::toString(Fw::StringBase&) const'
collect2: error: ld returned 1 exit status
/home/nathan/dev/fprime_math/fprime/mk/makefiles/Makefile:63: recipe for target '/home/nathan/dev/fprime_math/fprime/Ref/linux-linux-x86-debug-gnu-bin/Ref' failed
make[1]: *** [/home/nathan/dev/fprime_math/fprime/Ref/linux-linux-x86-debug-gnu-bin/Ref] Error 1
make[1]: Leaving directory '/home/nathan/dev/fprime_math/fprime/Ref'
Command exited with non-zero status 2
Build Time: 0:01.69
/home/nathan/dev/fprime_math/fprime/mk/makefiles/deployment_makefile.mk:48: recipe for target 'LINUX' failed
make: *** [LINUX] Error 2
```

and in this order : 
```Makefile
REF_MODULES := \
	Ref/Top \
	Ref/RecvBuffApp \
	Ref/SendBuffApp \
	Ref/SignalGen \
	Ref/PingReceiver \
	Ref/MathTypes \
        Ref/MathPorts \
	Ref/MathSender \
	Ref/MathReceiver 
```
I get this: 
```shell
Linking binary...
/home/nathan/dev/fprime_math/fprime/Ref/MathSender/linux-linux-x86-debug-gnu-bin/libRefMathSender.a(MathSenderComponentAc.o): In function `Ref::MathSenderComponentBase::set_mathOut_OutputPort(int, Ref::InputMathOpPort*)':
/home/nathan/dev/fprime_math/fprime/Ref/MathSender/MathSenderComponentAc.cpp:116: undefined reference to `Ref::OutputMathOpPort::addCallPort(Ref::InputMathOpPort*)'
/home/nathan/dev/fprime_math/fprime/Ref/MathSender/linux-linux-x86-debug-gnu-bin/libRefMathSender.a(MathSenderComponentAc.o): In function `Ref::MathSenderComponentBase::MathSenderComponentBase(char const*)':
/home/nathan/dev/fprime_math/fprime/Ref/MathSender/MathSenderComponentAc.cpp:281: undefined reference to `Ref::InputMathResultPort::InputMathResultPort()'
/home/nathan/dev/fprime_math/fprime/Ref/MathSender/MathSenderComponentAc.cpp:281: undefined reference to `Ref::OutputMathOpPort::OutputMathOpPort()'
/home/nathan/dev/fprime_math/fprime/Ref/MathSender/linux-linux-x86-debug-gnu-bin/libRefMathSender.a(MathSenderComponentAc.o): In function `Ref::MathSenderComponentBase::init(int, int)':
/home/nathan/dev/fprime_math/fprime/Ref/MathSender/MathSenderComponentAc.cpp:313: undefined reference to `Ref::InputMathResultPort::addCallComp(Fw::PassiveComponentBase*, void (*)(Fw::PassiveComponentBase*, int, float))'
/home/nathan/dev/fprime_math/fprime/Ref/MathSender/linux-linux-x86-debug-gnu-bin/libRefMathSender.a(MathSenderComponentAc.o): In function `Ref::MathSenderComponentBase::mathOut_out(int, float, float, Ref::MathOperation)':
/home/nathan/dev/fprime_math/fprime/Ref/MathSender/MathSenderComponentAc.cpp:542: undefined reference to `Ref::OutputMathOpPort::invoke(float, float, Ref::MathOperation)'
/home/nathan/dev/fprime_math/fprime/Ref/MathReceiver/linux-linux-x86-debug-gnu-bin/libRefMathReceiver.a(MathReceiverComponentImpl.o): In function `Ref::MathReceiverComponentImpl::mathIn_handler(int, float, float, Ref::MathOperation)':
/home/nathan/dev/fprime_math/fprime/Ref/MathReceiver/MathReceiverComponentImpl.cpp:73: undefined reference to `Ref::MathOp::MathOp()'
[... cut ...]
/home/nathan/dev/fprime_math/fprime/mk/makefiles/Makefile:63: recipe for target '/home/nathan/dev/fprime_math/fprime/Ref/linux-linux-x86-debug-gnu-bin/Ref' failed
make[1]: *** [/home/nathan/dev/fprime_math/fprime/Ref/linux-linux-x86-debug-gnu-bin/Ref] Error 1
make[1]: Leaving directory '/home/nathan/dev/fprime_math/fprime/Ref'
Command exited with non-zero status 2
Build Time: 0:01.63
/home/nathan/dev/fprime_math/fprime/mk/makefiles/deployment_makefile.mk:48: recipe for target 'LINUX' failed
make: *** [LINUX] Error 2
```

timcanham, 2019-02-27:
Try this order:

```
REF_MODULES := \
	Ref/Top \
	Ref/RecvBuffApp \
	Ref/SendBuffApp \
	Ref/SignalGen \
	Ref/PingReceiver \
	Ref/MathSender \
	Ref/MathReceiver \
        Ref/MathPorts \
	Ref/MathTypes
 ```

vigno88, 2019-02-27:
Wow, it works! How is the order working? Is it always : Component then Ports then types?

closed 2019-02-27 by vigno88

timcanham, 2019-02-27:
It's a link order issue, not an F` issue. Since they are archive files, if library1 depends on library2, library2 should be after library1 in the link order. For F', topologies depend on components, which depend on ports, which depend on types, so they have to be linked in that order.