nasa/fprime #25: Tutorial Updates

timcanham, 2019-03-02:
This issue is to track issues found while others are working their way through the tutorial.

timcanham, 2019-03-03:
1. Add `MathReceiver` header file to demo. Private members `m_factor1` and `m_factor1s` need to be shown.
2. Check link order in `modules.mk`
3. Add `Makefile` to each directory.

rdaruwala, 2019-03-10:
Hi Tim- I addressed a few of these issues in #15. The changes are currently merged into the dev branch, if you'd like I could open a new PR to merge those changes into master so that people new to fprime see the most up-to-date tutorial version

timcanham, 2019-03-12:
Fix event:

```
<event id="0" name="MS_COMMAND_RECV" severity="ACTIVITY_LO" format_string="Math Cmd Recvd: %f %d %f" >

to

<event id="0" name="MS_COMMAND_RECV" severity="ACTIVITY_LO" format_string="Math Cmd Recvd: %f %f %d" >
```
in Ref/MathSender/MathSenderComponentAi.xml

timcanham, 2019-03-28:
Add a comment that states that you can type `make` after the first `make rebuild`.

timcanham, 2019-05-08:
Fix variable initialization in `MathSender`.

AaronMcDaniel, 2019-06-25:
Ref/scripts/compile_ref_sequence.sh expects the seqgen.py file to be in Gds/bin/tkgui but it is actually in Gds/src/fprime_gds/tkgui/tools. Changing the path fixed my sequence compiling issues.

LeStarch, 2019-06-25:
@AaronMcDaniel can you submit this as a fix/PR? or direct me to the line that needs to be updated?

Also, thanks for reporting this.

AaronMcDaniel, 2019-06-27:
I opened PR #56 into devel that fixes the issue I mentioned above.

LeStarch, 2019-09-03:
PR #56 merged. Closing.

closed 2019-09-03 by LeStarch