nasa/fprime #27: make command cannot import cheetah

RyanPrintup, 2019-03-10:
Every time I attempt to build the demo, the make command fails to import cheetah.  `make gen_make` executed successfully.  Cheetah is also installed. Any ideas?

```
$ make
make[1]: Entering directory '/home/Ryan/fprime/Ref'
cd /home/Ryan/fprime/Fw/Buffer && mkdir -p /home/Ryan/fprime/Fw/Buffer/ac_dep && /usr/bin/python /home/Ryan/fprime/Autocoders/bin/codegen.py --build_root --default_topology_dict --dict_dir Dict BufferSerializableAi.xml --dependency-file /home/Ryan/fprime/Fw/Buffer/ac_dep/BufferSerializableAi.dep
Traceback (most recent call last):
  File "/home/Ryan/fprime/Autocoders/bin/codegen.py", line 49, in <module>
    from generators import GenFactory
  File "/home/Ryan/fprime/Autocoders/src/generators/GenFactory.py", line 28, in <module>
    from visitors import ComponentCppVisitor
  File "/home/Ryan/fprime/Autocoders/src/generators/visitors/ComponentCppVisitor.py", line 16, in <module>
    from generators.templates.component import cpp
  File "/home/Ryan/fprime/Autocoders/src/generators/templates/component/cpp.py", line 18, in <module>
    from Cheetah.Version import MinCompatibleVersion as RequiredCheetahVersion
ImportError: No module named Cheetah.Version
make[1]: *** [/home/Ryan/fprime/mk/makefiles/ac_targets.mk:1849: /home/Ryan/fprime/Fw/Buffer/BufferSerializableAc.cpp] Error 1
make[1]: Leaving directory '/home/Ryan/fprime/Ref'
Command exited with non-zero status 2
Build Time: 0:02.71
make: *** [/home/Ryan/fprime/mk/makefiles/deployment_makefile.mk:33: ac_lvl4] Error 2

```

timcanham, 2019-03-11:
This error usually means that Cheetah is not installed in the python version that is running doing the code generation. Do you have multiple installations of Python? Are you sure you installed Cheetah on the one used by the build?

ljreder, 2019-03-11:
Perhaps your using the wrong version.  There is a Python 3 version floating around.   We use version 2.4.4. and you can type the command `cheetah` and will get the following:

reder% cheetah
     __  ____________  __
     \ \/            \/ /
       \/    *   *     \/    CHEETAH 2.4.4 Command-Line Tool
         \      |       / 
           \  ==----==  /      by Tavis Rudd <tavis@damnsimple.com>
             \__________/       and Mike Orr <sluggoster@gmail.com>
              
    USAGE:
    ------
    cheetah compile [options] [FILES ...]     : Compile template definitions
    cheetah fill [options] [FILES ...]        : Fill template definitions
    cheetah help                              : Print this help message
    cheetah options                           : Print options help message
    cheetah test [options]                    : Run Cheetah's regression tests
                                                   : (same as for unittest)
    cheetah version                           : Print Cheetah version number

    You may abbreviate the command to the first letter; e.g., 'h' == 'help'.
    If FILES is a single "-", read standard input and write standard output.
    Run "cheetah options" for the list of valid options.

    *** USAGE ERROR ***: not enough command-line arguments

yashi, 2019-03-11:
I have seen this as well.

This could be the same problem with #26.  Are you using `/usr/bin/python`?  As you can see from the log, the build system is choosing `/usr/bin/python`.  If you have Cheetah installed for `/usr/bin/python` but still doesn't work, check environment variables when the command in the log is invoked.  The build system might be playing with it.

I'll check my env again when I have time but meanwhile please investigate it.

RyanPrintup, 2019-03-16:
I ended up ditching the Cygwin installation in favor of a Ubuntu VM. Everything is working as expected

closed 2019-03-16 by RyanPrintup