nasa/fprime #28: Custom Component Questions

brhs17, 2019-03-15:
I've started trying to make my own component based on the math tutorial component, but have gotten confused by the names of seemingly similar sections of code:
`<enum name="MathOp">`
`<enum name="Operation">`
`<enum name="MathOperation">`

Also, Is there a way to have components and ports two layers below Ref?  I tried by putting a mod.mk file in the directory in Ref that had `SUBDIRs = nextLayer`, but this didn't work.

brhs17, 2019-03-15:
Is there a list of tasks that need to be done to make a component?  This would be very helpful so we could just work down the list instead of trying to get an implied list from the math tutorial.

brhs17, 2019-03-28:
What do the `0` and `10` arguments mean here:
[image]

timcanham, 2019-03-28:
`0` is the instance of the component if you have more than one (see the `init()` function of the component). For unit tests, it can always be `0`, unless you do different things based on instance. The `10` is a command sequence number you can pick to make sure your code returns the correct one in the command response.

Azlinkix, 2019-04-25:
Hi, I'm Sarah Sposato and I am also working on the math components tutorial.
I am running into an error with "make impl" 

```
make: *** No rule to make target `impl'.  Stop.

````

All of the other steps have been somewhat successful. 
I'm still assuming I am in Ref.
I don't see a makefile in Ref for implementation?

rdaruwala, 2019-04-25:
Hi Sarah -- You'll want to run `make impl` from within a component directory to generate the implementation files, not from general Ref application folder. Each component directory requires its own Makefile following the format specified in the tutorial.

Azlinkix, 2019-04-25:
Ah I see. Thank you for explaining that!
So if the tutorial says "To generate a set of files for testing, from the **module directory type**: 'testcomp'---to me this is saying to go into fprime/Md (modules?)

I was getting a similiar "make: *** No rule to make target `testcomp'.  Stop." So I figured I was in the wrong place?

LeStarch, 2019-04-25:
It is similar to the above comment. Running `make impl` and `make testcomp` both should be done in the directory of the Component, Port, or Topology you are working on.  These are 'Modules'.

Assuming you put MathSender under 'Ref':
```
cd fprime/Ref/MathSender
make impl
make test_comp
```

We will update the tutorial to make it more clear.

LeStarch, 2019-04-25:
**Note:** In order for these to run properly, the mk/config/modules/modules.mk must be updated to add the path `Ref/MathSender` and `make gen_make` should be rerun.

timcanham, 2019-04-25:
Also note that since `make impl` and `make testcomp` are component specific targets, they only work in directories that have component XML.

LeStarch, 2019-09-03:
Last updated in April. Closing.

closed 2019-09-03 by LeStarch