nasa/fprime #31: No Class Prototype?

brhs17, 2019-03-22:
I got this error when trying to rebuild a folder in which I had just run `make impl` on successfully.
[image]

timcanham, 2019-03-23:
It's hard to see what's happening. Is this on a branch.

brhs17, 2019-03-23:
Sorry for the delayed response, I put what I currently have on my computer here:
https://github.com/brhs17/fprime/tree/MOpAgg
I messed with a bunch of stuff, but still get similar errors.

timcanham, 2019-03-24:
First issue:

```xml
		<event id="104" name="CLEAgg_REQUEST_COMPLETED" severity="ACITIVITY_HI" format_string="Oparation Performed %s" >
		<args>
		<arg nam="val" type="Ref::DataRequest"></arg>
		</args>
		</event>
```

`nam` should be `name`, and `ACITIVITY_HI` should be `ACTIVITY_HI`. Sometimes the XML validator script can't tell you where the error is, so you have to use the process of elimination.

```c++
In file included from /mnt/c/data/source3/fprime/forks/brhs17/fprime/Ref/CLEAgg/CLEAggComponentAc.hpp:48:0,
                 from /mnt/c/data/source3/fprime/forks/brhs17/fprime/Ref/CLEAgg/CLEAggComponentAc.cpp:22:
/mnt/c/data/source3/fprime/forks/brhs17/fprime/Ref/CLETypes/CLERequestSerializableAc.hpp:25:9: error: redeclaration of ‘GyroX’
         GyroX,
         ^
In file included from /mnt/c/data/source3/fprime/forks/brhs17/fprime/Ref/CLEAgg/CLEAggComponentAc.hpp:36:0,
                 from /mnt/c/data/source3/fprime/forks/brhs17/fprime/Ref/CLEAgg/CLEAggComponentAc.cpp:22:
/mnt/c/data/source3/fprime/forks/brhs17/fprime/Ref/CLEPorts/DataRequestPortAc.hpp:25:9: note: previous declaration ‘Ref::DataRequestPortZ GyroX’
         GyroX,
         ^
In file included from /mnt/c/data/source3/fprime/forks/brhs17/fprime/Ref/CLEAgg/CLEAggComponentAc.hpp:48:0,
                 from /mnt/c/data/source3/fprime/forks/brhs17/fprime/Ref/CLEAgg/CLEAggComponentAc.cpp:22:
/mnt/c/data/source3/fprime/forks/brhs17/fprime/Ref/CLETypes/CLERequestSerializableAc.hpp:26:9: error: redeclaration of ‘GyroY’
         GyroY,
         ^
In file included from /mnt/c/data/source3/fprime/forks/brhs17/fprime/Ref/CLEAgg/CLEAggComponentAc.hpp:36:0,
                 from /mnt/c/data/source3/fprime/forks/brhs17/fprime/Ref/CLEAgg/CLEAggComponentAc.cpp:22:
/mnt/c/data/source3/fprime/forks/brhs17/fprime/Ref/CLEPorts/DataRequestPortAc.hpp:26:9: note: previous declaration ‘Ref::DataRequestPortZ GyroY’
         GyroY,
         ^
In file included from /mnt/c/data/source3/fprime/forks/brhs17/fprime/Ref/CLEAgg/CLEAggComponentAc.hpp:48:0,
                 from /mnt/c/data/source3/fprime/forks/brhs17/fprime/Ref/CLEAgg/CLEAggComponentAc.cpp:22:
/mnt/c/data/source3/fprime/forks/brhs17/fprime/Ref/CLETypes/CLERequestSerializableAc.hpp:27:9: error: redeclaration of ‘GyroZ’
         GyroZ,
         ^
In file included from /mnt/c/data/source3/fprime/forks/brhs17/fprime/Ref/CLEAgg/CLEAggComponentAc.hpp:36:0,
                 from /mnt/c/data/source3/fprime/forks/brhs17/fprime/Ref/CLEAgg/CLEAggComponentAc.cpp:22:
/mnt/c/data/source3/fprime/forks/brhs17/fprime/Ref/CLEPorts/DataRequestPortAc.hpp:27:9: note: previous declaration ‘Ref::DataRequestPortZ GyroZ’
         GyroZ,
         ^
In file included from /mnt/c/data/source3/fprime/forks/brhs17/fprime/Ref/CLEAgg/CLEAggComponentAc.hpp:48:0,
                 from /mnt/c/data/source3/fprime/forks/brhs17/fprime/Ref/CLEAgg/CLEAggComponentAc.cpp:22:
/mnt/c/data/source3/fprime/forks/brhs17/fprime/Ref/CLETypes/CLERequestSerializableAc.hpp:28:9: error: redeclaration of ‘RTCTime’
         RTCTime,
         ^
In file included from /mnt/c/data/source3/fprime/forks/brhs17/fprime/Ref/CLEAgg/CLEAggComponentAc.hpp:36:0,
                 from /mnt/c/data/source3/fprime/forks/brhs17/fprime/Ref/CLEAgg/CLEAggComponentAc.cpp:22:
/mnt/c/data/source3/fprime/forks/brhs17/fprime/Ref/CLEPorts/DataRequestPortAc.hpp:28:9: note: previous declaration ‘Ref::DataRequestPortZ RTCTime’
         RTCTime,
         ^

```
This one is a C++ violation - you have `Gyro_*` and `RTCTime` enumeration members declared in two places in the same namespace, in `DataRequestPort` and `CLERequestSerializable`.

Once I fixed those two in my clone, the component compiled okay.

brhs17, 2019-03-24:
I think I implemented your fixes for CLEAgg, but I still get this error when I try to `make rebuild` or `make impl` in the CLEAgg directory:
[image]

Additionally, when I try to `make rebuild`  in the CLEMOps directory, I get this error, which seems completely different:
[image]

I uploaded the changes onto the branch

rdaruwala, 2019-03-24:
To fix your `AssertionError: Extra element parameters in interleave` error, remove 

```
	<parameters>
	</parameters>
```

from your XML file.

For your second error, my best guess is that those files were generated using an outdated/incorrect XML file. If you delete & remake the Impl cpp/hpp files, they contain `DataRequest data` instead of `options data`, which is what's specified by your XML file

brhs17, 2019-03-28:
Is there a faster way to compile our code when modifying c files than `make rebuild`?  It seems excessive to wait about 30s just to get errors from one file.

timcanham, 2019-03-28:
You don't have to do a rebuild; just `make` from the `Ref` directory will compile the changed files.

brhs17, 2019-03-28:
What do I need to do if I realize I made a mistake in an XML file and I have started working on the c files?  I named my port items the same as something else, which now causes a collision.

brhs17, 2019-03-28:
I edited the port xml file, ran `make gen_make` in the port directory, and then it compiled.  Is there anything else I should do?

timcanham, 2019-03-28:
What I've done in the past is to re-run the `make impl` target to regenerate the stubs, then copy the new functions from the stubs into my implementation `.cpp` and `.h`.

timcanham, 2019-03-28:
You can just recompile the port files; that's all you have to do.

brhs17, 2019-03-28:
Why is this 0 required and where does it come from?  I had to also put it in my own component.
[image]

[image]

timcanham, 2019-03-28:
The `0` is the port instance. All ports are arrays of the same port type. If you don't specify `max_number` in the port declaration in your component, it will default to `1`, but you still have to give `0` as the instance in the port call. The port arrays allow you to connect to multiple components that you would invoke in the same way. You can call `this->getNum_<PortName>_OutputPorts()` if you want your code to automatically scale to the number of ports.

closed 2019-03-28 by brhs17