nasa/fprime #32: General Errors Issue

brhs17, 2019-03-28:
I am working on the testing section of my CLEMOps component, and I get this error when I try to run `make ut`.  I haven't started working on the tests themselves, but I think I have everything setup, including the blank testing functions.
https://github.com/brhs17/fprime/tree/MOpAgg
[image]

timcanham, 2019-03-28:
Whenever you get the "cannot declare .... abstract type", it means that a pure virtual function from the base class has not been implemented. You need to make sure all the pure virtual functions, which are handlers, in the base class are implemented in your derived testing class.

brhs17, 2019-03-30:
I don't know how to do that, both the handlers in my `ComponentImpl.cpp` file look implemented.  The error tells me that the `OPS_Get_Data_cmdHandler` in 'ComponentAc.hpp` is not implemented, but I don't know where I'm supposed to write that, or really what the code needs to do.

timcanham, 2019-03-31:
You write the function in the `ComponentImpl.cpp`. A similar command handler can be found here:

https://github.com/nasa/fprime/blob/master/docs/Tutorials/MathComponent/MathSender/MathSenderComponentImpl.cpp#L76

brhs17, 2019-03-31:
What did I do wrong in my implementation?
[image]
[image]

timcanham, 2019-03-31:
Looks like virtual is `OPS`, yours is `Ops`. You have to get the spelling exactly right.

brhs17, 2019-03-31:
ok, I just needed to run the impl command to make new files and copy and paste everything.

timcanham, 2019-03-31:
I've been using Fprime for a while now, and that's how I do it. Avoids errors like this.

brhs17, 2019-04-01:
I'm getting a bunch of undefined Reference errors, what should I be looking for to fix them?
[image]

timcanham, 2019-04-01:
Check the `mod.mk` for the unit test. The `TEST_MODS` variable should have any modules your component relies on for testing.

brhs17, 2019-04-02:
I'm not sure where this error came from, I'm not even sure how its related to something I write:
[image]

timcanham, 2019-04-02:
The `Fw::ParamValid` type is used to see if parameters successfully loaded. If you haven't defined any parameters in your XML, the headers to define it aren't included. If you added some parameters and then removed them all, you could see this error.

brhs17, 2019-04-02:
What should I do to fix these warnings?
[image]

brhs17, 2019-04-03:
I'm working on the topology and getting an error I don't understand at all:
[image]

LeStarch, 2019-04-04:
Your constructor is expecting a name when you declare the component.  Try something like this:

```
Ref::CLEAggComponentImpl("CLE Agg");
```

LeStarch, 2019-04-04:
In response to your noted warnings 2 comments ago: these warnings occur when the order you define your member variables in the *.hpp are in a different order than the order in which you initialize them in the initializer list in the *.cpp.

The best way to fix those warnings is to go through your *.hpp in and list all declared members in-order of declaration from top to bottom.  Then go through the initializer list in th *.cpp and reorder it to match.

brhs17, 2019-04-16:
I'm trying to run the ground system with my new components in addition to the math components and am getting this error, which is strange because the math components worked before.
[image]

rdaruwala, 2019-04-16:
Make sure to run `make dict_install` before opening your GUI application

LeStarch, 2019-04-16:
Right, if you do not build with `make rebuild` you need to ensure you run `make dict_install` since the dictionaries can get out-of-date, or cleared.

brhs17, 2019-04-21:
I have my components mostly working, but I get this error whenever I try to use them that seems to be related to the text they are supposed to return somewhere, however, the error message seems to be completely void of any identifying information of where is it coming from.  The `Ref Application` window displays everything as expected, but nothing shows up in the `Log Events` tab, which is bad.
[image]

rdaruwala, 2019-04-21:
This looks to be the exact same error you encountered in #22. The fix will likely be along the same lines. Check the XML file to make sure you're passing in the right types of arguments in the right order

brhs17, 2019-04-21:
Ok, so I seemed to have fixed it by changing something to want an integer instead of a float, but this seems super confusing because a string is mentioned in the error message.  Also, why did it not get autocasted since getting an integer and converting it to a float doesn't lose precision.

timcanham, 2019-04-22:
This is a ground software side issue that we need to fix to give a better error message.  As @rdaruwala says, it indicates a mismatch between the format specifiers in the event/telemetry message and the actual type of the argument.

Azlinkix, 2019-05-03:
Hi,

I'm running into a similar issue with the make ut command except the error is:

[image]

Azlinkix, 2019-05-03:
Running the following command "make ut run_ut" returns the same error.

Running "make ut run_ut cov" returns a similar error:

[image]

Any suggestions on how I might move forward?

timcanham, 2019-05-03:
It looks like `main()` isn't defined anywhere in your test files. Do you have this file in your unit test `mod.mk`?

https://github.com/nasa/fprime/blob/master/docs/Tutorials/MathComponent/MathSender/test/ut/main.cpp

Azlinkix, 2019-05-07:
Thanks for the tip--turns out I missed a file.

Almost done with the tutorial! 

One (hopefully last) question:

[image]

I got alot of syntax errors? Or all of these are unititated? I looked at the reference files under Ref > Docs> Tutorials > MathTutorials and my files are the same.

Any suggestions or needed clarifications?

brianthespoon, 2019-05-07:
Also stumped by this one

timcanham, 2019-05-08:
For `m_factor`` and `m_factor1s`, make sure they are declared as private data members of your class:

```c++
      // stored factor1
      F32 m_factor1;
      // number of times factor1 has been written
      U32 m_factor1s;
```

See https://github.com/nasa/fprime/blob/master/docs/Tutorials/MathComponent/MathReceiver/MathReceiverComponentImpl.hpp#L101.

The header file is missing from the tutorial and should be added in. We have added it to #25.

timcanham, 2019-05-08:
The uninitialized variable warning looks genuine. Maybe you are using a new compiler that checks for it. We'll add a fix to the tutorial, but in the meantime just initialize the variables where you declare them.

brh68, 2019-06-08:
I am trying to add two components the the RPI topology, and keep getting this error:
[image]
I am mostly worried about why aggregator isn't working because RTCComponent doesn't compile yet.

timcanham, 2019-06-09:
It seems like there could be a typo in the name or namespace. Is this on a branch somewhere?

brh68, 2019-06-11:
I'm trying to run a modified version of `Ref`, but I get some fatal 342 error after everything opens up, but before doing anything.
[image]

LeStarch, 2019-06-12:
It looks like you need to increase the configuration for the command dispatcher.

In `Fw/Cfg/AcConstants.ini`, what is the line "CmdDispatcherComponentCommandPorts" set to?
```
CmdDispatcherComponentCommandPorts  =       30
```
Is it 30 like above?

LeStarch, 2019-06-12:
This may also be an issue in your Topology.  In Your Topology Ai file, you should look for lines with the `cmdDisp`  where the number at the end is 100, or 101.  You need to keep these numbers sequential.
```
         <target component = "cmdDisp" port = "compCmdReg" type = "CmdReg" num = "11"/>
```

brh68, 2019-06-12:
Thank you, changing the `cmdDisp` numbers to be sequential fixed the problem.

LeStarch, 2019-06-12:
If you have more than 30 components connected to the cmdDisp, you need to increase the above config.

LeStarch, 2020-03-03:
I am closing this issue, as I see no updates.  Please feel free to continue discussion on our mailing list:  https://groups.google.com/forum/#!forum/fprime-community

Or open a new issue if you detect a new bug.

closed 2020-03-03 by LeStarch