nasa/fprime #33: make impl not generating template files

RyanPrintup, 2019-04-12:
I am trying to implement a component in FPrime by following the getting started tutorial.  I have made the Ai.xml, Events.xml, Makefile, and mod.mk.  Currently, I am trying to generate implementation templates.  Running `make impl` results in the following console output:
'Generating implementation template for DrvLinuxI2CDriver
make[1]: Entering directory '/home/ubnl-fprime/fprime/Drv/LinuxI2CDriver'
cd /home/ubnl-fprime/fprime/Drv/LinuxI2CDriver && /usr/bin/python /home/ubnl-fprime/fprime/Autocoders/bin/codegen.py --build_root -t 
Usage: /home/ubnl-fprime/fprime/Autocoders/bin/codegen.py [options] xml_filename
make[1]: Leaving directory '/home/ubnl-fprime/fprime/Drv/LinuxI2CDriver'
Build Time: 0:00.69'

Makefile:
`MODULE_DIR = Drv/LinuxI2CDriver
MODULE = $(subst /,,$(MODULE_DIR))

BUILD_ROOT ?= $(subst /$(MODULE_DIR),,$(CURDIR))
export BUILD_ROOT

include $(BUILD_ROOT)/mk/makefiles/module_targets.mk`

mod.mk:
`MODULE_DIR = Drv/LinuxI2CDriver
MODULE = $(subst /,,$(MODULE_DIR))

BUILD_ROOT ?= $(subst /$(MODULE_DIR),,$(CURDIR))
export BUILD_ROOT

include $(BUILD_ROOT)/mk/makefiles/module_targets.mk`

LeStarch, 2019-04-12:
Right away, I can see that your mod.mk file contains the wrong information.  It should contain as source and header definition like this:

** mod.mk **
```
SRC = LinuxI2CDriverAi.xml

HDR = 
```

Of course, replace your SRC definition with the name of your own Ai.xml, if I guessed incorrectly.

LeStarch, 2019-04-12:
Also, could you put a link here to the tutorial you are following, so I can ensure this information is updated?

RyanPrintup, 2019-04-12:
I accidentally copied my Makefile twice. I updated my original post with the correct mod.mk information.  This is the tutorial that I've been following: https://nasa.github.io/fprime/getting-started.html

RyanPrintup, 2019-04-12:
I reviewed my code again and discovered errors in my xml syntax.  Issue resolved.

RyanPrintup, 2019-05-29:
I am continuing to have this error, but with a different console output.

```
Generating implementation template for 
make[1]: Entering directory '/home/ubnl-fprime/rogue-zero/Balloon/BurnWire'
make[1]: *** No rule to make target '_impl'.  Stop.
make[1]: Leaving directory '/home/ubnl-fprime/rogue-zero/Balloon/BurnWire'
Command exited with non-zero status 2
Build Time: 0:00.51
/home/ubnl-fprime/rogue-zero/mk/makefiles/module_targets.mk:81: recipe for target 'impl' failed
make: *** [impl] Error 2
```

LeStarch, 2019-05-31:
It looks like the Makefile isn't specifying the directory path correctly.  Also, did you run `make gen_make` before running `make impl`?

LeStarch, 2020-03-03:
I'd recommend looking into CMake, as that is what we intend to do in the future.

I am closing this issue, as I see no updates.  Please feel free to continue discussion on our mailing list:  https://groups.google.com/forum/#!forum/fprime-community

Or open a new issue if you detect a new bug.

closed 2020-03-03 by LeStarch