nasa/fprime #34: Ref app in master branch is not building on Mac OSX Mojave

ljreder, 2019-04-22:
I attempted to clone the master branch and build on my new MAC running Mojave.  It looks like the Mac is missing an ssl library.  When making the Ref app I get this at the end:

Linking binary...
ld: library not found for -lssl
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [/Users/reder/Documents/Workspace/FP2/fprime/Ref/darwin-darwin-x86-debug-llvm-bin/Ref] Error 1
        5.72 real        11.34 user         6.30 sys
make: *** [DARWIN] Error 2

Also it appears the Make system is not setting the PYTHONPATH for the autocoder.

rdaruwala, 2019-04-23:
@ljreder I was able to fix the SSL issue when building on my MacBook by running the following commands:

```
xcode-select --install
export LIBRARY_PATH=$LIBRARY_PATH:/usr/local/opt/openssl/lib/
```

ljreder, 2019-04-23:
Thanks Rohan, but actually I did not have the path to openssl you specify.  
However I was able to install the libraries using MAC ports and it seems to work now.
I simply did this

xcode-select --install
sudo port install openssl
setenv LIBRARY_PATH /opt/local/lib

I think the xcode step just updates versions.

Recommend this issue closed.   Also would be useful to update cmake to handle this.

closed 2019-04-23 by ljreder