nasa/fprime #38: Issue when running runBoth.sh

ecruz480, 2019-05-18:
I have successfully ran `make rebuild_rpi` but am running into issues when trying to launch the program itself with `./runBoth.sh`. 
[https://imgur.com/zIuy5we](Photo with runBoth issue). Any help is appreciated, currently running on RPI 3b+ with Raspian 4.14 (Debian Stretch)

timcanham, 2019-05-20:
It looks like a bug in the scripts when we converted to sh from tcsh:

```
if !($?BUILD_ROOT) then
    set curdir = "${PWD}"
    setenv BUILD_ROOT `dirname $0`/../..
    cd $BUILD_ROOT
    setenv BUILD_ROOT ${PWD}
    cd ${curdir}
endif
```

It should be something like:

```
if !($?BUILD_ROOT) then
    set curdir = "${PWD}"
    export BUILD_ROOT=`dirname $0`/../..
    cd $BUILD_ROOT
    export BUILD_ROOT=${PWD}
    cd ${curdir}
endif
```

Next time I get on a Pi, I'll try it, but perhaps you could try it for us.

timcanham, 2019-05-20:
Note that sometimes the GUI has issues running on the Pi even if the scripts are working and it can be sloooooow. The alternate pattern is to run the GUI on a Linux machine (`runGui.sh`) then run the app on the Pi (`runRPI.sh` or `runRPICross.sh` if you cross-compiled). You will have to edit the two scripts and change the IP addresses to whatever network you are running on.

ecruz480, 2019-05-20:
Yes! Using `export` cleared up my prior issues but I am now stuck on python's importing tools.

[image]

Just from reading a few posts online, it might be an issue with `sys.path` and how it is initialized by the program. I am currently running python 3.5.3 but I seems like there are still problems with how the system interprets what is a package directory and the careful use of `__int__.py`. I am an electrical engineer by trade and heart so my usefulness is trying to catch up to my eagerness, but I plan on trying to cross-compile it soon.

[Link on import issue.](http://python-notes.curiousefficiency.org/en/latest/python_concepts/import_traps.html#the-init-py-trap)

LeStarch, 2019-05-20:
Yeah, the Gse/Gds layer was merged together, however; the "runBoth" script was not updated to reflect this.

```
export PYTHONPATH="${BUILD_ROOT}/Gse/src:${BUILD_ROOT}/Gse/generated/FlightDemo/serializable"
```

Should read something like:
```
export PYTHONPATH="${BUILD_ROOT}/Gds/src:${BUILD_ROOT}/RPI/py_dict/serializable"
```

LeStarch, 2019-05-31:
@timcanham can we close this, or do we need to create a run-on-rpi wrapper?

closed 2019-09-03 by LeStarch