nasa/fprime #47: Issue with installations for RPI Demo

aaronbuchwald, 2019-06-07:
Hi,

I'm trying to do the RPI demo, but I'm ran into problems with installation.

When I ran ./mk/os-pkg/ubuntu-packages.sh it works until it throws the error that it cannot locate libc-i386. I think this is the reason that the next two installations fail the first time.

For the first one:
sudo pip install -r mk/python/pip_required_build.txt

I am getting the error
/bin/sh: 1: xslt-config: not found multiple times

And then later it says to make sure that the development packages of libxml2 and libxslt are installed.

I'm also getting the error that the Python.h and libxml/xpath.h files are not included.

To fix this, I used apt-get to get what was missing.

sudo apt-get install python-dev
sudo apt-get install python-lxml

This allows sudo pip install -r mk/python/pip_required_build.txt to run to completion with no errors.

For the second one, mk/python/pip_required_gui.txt the requirements file simply does not exist, so I guess the README should be updated or an empty text file added so that it doesn't throw an error when installing and requirements can be added later if anything becomes necessary for the gui.

Now when I run
make rebuild_rpi
It results in the warning that I I don't have the c version of NameMapper installed each time the make command tries to use the cheetah compiler.
Then the error that:
/usr/bin/time: not found
terminates the make command.

I did this:
which time
This is unable to find the time, which means that running ubuntu_packages.sh (which should have installed it) must not have successfully installed time for some reason. I ran this directly from the command line doing:
sudo apt-get install time.
After this I was able to successfully run make rebuild_rpi.

Does anyone know why the ubuntu_packages.sh file is not able to locate libc6-i386? I ran sudo apt-get install libc6-i386 and it gave me the same error that it couldn't find it, which I think was the cause of other dependencies failing to install and the rest of the problems. Is libc6-i386 necessary for some other reason?

timcanham, 2019-06-08:
What version of Ubuntu did you try to install to?

aaronbuchwald, 2019-06-10:
Linux raspberry pi 4.1.18

LeStarch, 2019-06-10:
Are you trying to build and run on the RPI? or are you building on a computer, and uploading and running on the RPI?

aaronbuchwald, 2019-06-10:
I have been trying to build and run on the RPI, should it be vice versa?

LeStarch, 2020-03-03:
Checkout CMake (perhaps you are already).

I am closing this issue, as I see no updates.  Please feel free to continue discussion on our mailing list:  https://groups.google.com/forum/#!forum/fprime-community

Or open a new issue if you detect a new bug.

closed 2020-03-03 by LeStarch