nasa/fprime #48: Port with Return Type

brh68, 2019-06-08:
I am trying to make a port that has a return type, and have put the return type in the port definition XML.  However, when I make a component that uses this port and should return a value, the impl files are generated with the associated function having a return type of void.  This causes the compile error below when I rebuild, and makes sense since there doesn't seem to be a way signify a return type in component XML.
[image]
Do I have to manually chance the return types of the Impl files, or is there something else I can do to fix this through the autocoder?

timcanham, 2019-06-09:
What type of port did you define it as? (sycn, async, guarded?)

brh68, 2019-06-09:
Synchronous Input

brh68, 2019-06-10:
I put it in a repository after moving things around, so the errors have changed slightly:
[image]
Here is the repository:
https://github.com/brh68/Cislunar-Flight-Software/tree/peripherals

LeStarch, 2019-06-10:
Did you remake `make impl` and then copy over the new template files?

brh68, 2019-06-10:
The problem remains after I rerun `make impl`.

brh68, 2019-06-11:
I manually changed the return types of the functions in the impl files, which seems to work.

LeStarch, 2019-06-12:
We'll still need to test for this bug before closing out the issue.  Thanks for the report.

LeStarch, 2020-03-03:
Reviewing this issue, this is expected behavior.

We don't automatically update the impl files, as we do not want to clobber outstanding work that the user has done.

Once the implementation generator is run, the user owns the files and must manually update or manually regenerate (and move from the _template outputs).

closed 2020-03-03 by LeStarch