nasa/fprime #49: Setting fprime as an upstream branch for our repo

aaronbuchwald, 2019-06-10:
Hello, we forked the F Prime repository a while back in order to create our repository and want to merge our repo with all of the edits since then. How can we set the original f prime repo as an upstream branch and which branch should we rebase with? Is it reasonably unlikely that if we rebase with the newest updates it would cause breaking changes?

LeStarch, 2019-06-10:
To set an upstream to the nasa repo:
```
git remote add nasa-upstream http://github.com/nasa/fprime.git
```

Then you can use that upstream with commands like `fetch`
```
git fetch nasa-upstream
```

LeStarch, 2019-06-10:
If you are worried about "breaking changes" try it in a throwaway branch first, and see how bad the changes are:

```
git checkout <your-branch>
git checkout -b "throwaway-1"
git fetch nasa-upstream
git merge nasa-upstream/devel
```

LeStarch, 2019-06-10:
I don't believe you have lots and lots of conflicts, unless you are editing core F´ files.

aaronbuchwald, 2019-06-10:
Thank you so much

closed 2019-06-10 by aaronbuchwald