nasa/fprime #51: build errors using make gen_make

wr3484, 2019-06-10:
I am having trouble with a gps component we are trying to create, the make gen_make command is providing an error that I cannot figure out. This is the error obtained:
[image]

My code does not even reference the svc directory so I am not really sure how to fix that.

Here is the makefile for it:
[image]

LeStarch, 2019-06-10:
It looks like "BufferAccumulator" is referenced in your `mk/configs/modules/modules.mk` file.  Which branch did you start from?  Are you on the latest NASA devel?

wr3484, 2019-07-06:
We are working on a side project incorporating fprime into a weather balloon, simple ideas, gps, etc.  I scrapped the previous idea, made a clone of fprime directly and copied our Balloon file into the fprime directory. The `make gen_make`  worked, but when I actually try to build using the `make` command, there is a strange error about Linux. Any suggestions? 

[image]

LeStarch, 2019-07-06:
Is this compiling a Topology? If so, make sure you have defined a `main` function.

wr3484, 2019-07-08:
It is compiling a topology and we do have a main function. It builds on a separate computer running a VM, but I am attempting to build through a Linux operating system. 

[image]
[image]

LeStarch, 2019-07-08:
Have you ensured that the "Topology.cpp" is in your mod.mk file, and that you have refreshed with `make gen_make`?

wr3484, 2019-07-08:
Yes, it is included in the mod.mk file and refreshed with a `make gen_make`

[image]

LeStarch, 2020-03-03:
Checkout the new CMake build system.

I am closing this issue, as I see no updates.  Please feel free to continue discussion on our mailing list:  https://groups.google.com/forum/#!forum/fprime-community

Or open a new issue if you detect a new bug.

closed 2020-03-03 by LeStarch