nasa/fprime #55: Development-Process.md / issue

as2756, 2019-06-24:
Hello, 
I am trying to set up a port using the directions found at: https://github.com/fprime-community/fprime-course-materials/blob/master/Flight_Software_Workshop-June_2019/Development-Process.md
However, I am having difficulties once I arrive to step 7 (in the port directory: make to autocode the port). 

I am getting an error that "status" was not declared in the scope (screenshots of the error are attached below). I have talked to a couple of my  teammates and they have had the same issue. Does anyone have any ideas what I should do to fix the problem? Thank you!

[image]
[image]

aaronbuchwald, 2019-07-08:
Hello, I saw this issue and wasn't sure what to make of it. The error seems to be coming from code that was autogenerated with the word status when it maybe should have written static as in the suggested alternative. Any idea if this is an issue with the autocoders or if it was caused by some error in how the component was written? Would it be helpful to share the code for the component that caused this error?

timcanham, 2019-07-08:
Do you have a branch we could look at?

LeStarch, 2020-03-03:
I am closing this issue, as I see no updates.  Please feel free to continue discussion on our mailing list:  https://groups.google.com/forum/#!forum/fprime-community

Or open a new issue if you detect a new bug.

closed 2020-03-03 by LeStarch