nasa/fprime #57: Connecting RPI to server port

aaronbuchwald, 2019-07-02:
Hello,

I am trying to connect the RPI to my linux machine and I am running into a strange bug. I was running the run_rpi_cross.sh file on the linux machine and it was booting up and launching the GUI, but then when I ran the RPI binary on the raspberry pi it gives me the message that it has connected to the server on port 50000, but the linux machine itself does not recognize or acknowledge the connection. I tried running the RPI file without it running on the linux machine and it still gives me the message that it has connected to the linux machine. Any idea why this is occurring or if this is a bug?

LeStarch, 2019-07-02:
Check to see if the threaded tcp server is orphaned.

```
ps -f | grep -i tcpserver.py 
```

aaronbuchwald, 2019-07-08:
The problem seems to have gone away and I'm unable to replicate it. Is this an issue that you've seen before? If it occurs again, how would you go about fixing it?

LeStarch, 2019-07-08:
If it occurs again, make sure all your F´ ground system python processes are shut down.  Then rerun.

aaronbuchwald, 2019-07-08:
Awesome and thank you very much. Closing this issue then.

closed 2019-07-08 by aaronbuchwald