nasa/fprime #59: F Prime Initialization - CMake and Sample Application Start up Problems

msyed-aubbie, 2019-07-22:
Hello,

We are just starting f Prime exploration. In order to do so, the first steps were to download f prime on Windows and Linux machines to be compiled. I have not been able to build/ compile the program yet. Requesting your feedback in the following scenarios.

**Windows Problem(s)** 

1. Downloaded fPrime master branch using Eclipse which created fprime-sw project:
Error 1 a - Console Output:
`01:20:38 **** Build of configuration Default for project fprime-sw ****
make gen_make all 
Generating templates
make[1]: Entering directory '/c/Users/git/fprime/Autocoders/Python/src/fprime_ac/generators/templates/component'
make[1]: /bin/cheetah-compile: Command not found
/bin/cheetah-compile cpp.tmpl
make[1]: *** [cpp.py] Error 127
make[1]: Leaving directory '/c/Users/git/fprime/Autocoders/Python/src/fprime_ac/generators/templates/component'
make: *** [compile-templates] Error 2
"make gen_make all" terminated with exit code 2. Build might be incomplete.`
Error 1 b:
Download also created a 'MagicDrawCompPlugin' project that leads to the following error when built:
`package gov.nasa.jpl.componentaction;`
The type java.lang.Object cannot be resolved. It is indirectly referenced from required .class files. 

2. CMake on Cygwin Error:

 cmake ../Ref/
 CMakeLists.txt:27 (message):
  [F-PRIME] No platform config for 'Windows'.  Please create:
  'C:/cygwin/home/fprime/cmake/platform/Windows.cmake'

Call Stack (most recent call first):
  C:/cygwin/home/fprime/cmake/FPrime.cmake:28 (include)
  CMakeLists.txt:32 (include)

3. Sample Application Error on Cygwin

` ~/fprime/Ref
$ make gen_make
Generating templates
make[1]: Entering directory '/c/cygwin/home/fprime/Autocoders/Python/src/fprime_ac/generators/templates/component'
/bin/cheetah-compile cpp.tmpl
make[1]: /bin/cheetah-compile: Command not found
make[1]: *** [cpp.py] Error 127
make[1]: Leaving directory '/c/cygwin/home/fprime/Autocoders/Python/src/fprime_ac/generators/templates/component'
make: *** [compile-templates] Error 2
`

**Linux Problem(s)**

4. Cmake Error on Linux:

`cmake ../Ref/
-- The C compiler identification is GNU 4.4.7
-- The CXX compiler identification is GNU 4.4.7
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- F' core directory set to: /home/mas0047/fprime/fprime/cmake/..
-- [VALIDATION] Running: python;/home/mas0047/fprime/fprime/cmake/support/validation/pipsetup.py
enum34 is missing.
Markdown is missing.
Traceback (most recent call last):
  File "/home/mas0047/fprime/fprime/cmake/support/validation/pipsetup.py", line 40, in <module>
    sys.exit(0 if validate_python_setup() else 1)
  File "/home/mas0047/fprime/fprime/cmake/support/validation/pipsetup.py", line 35, in validate_python_setup
    except pkg_resources.RequirementParseError:
AttributeError: 'module' object has no attribute 'RequirementParseError'
CMake Error at /home/mas0047/fprime/fprime/cmake/support/validation/Validation.cmake:38 (message):
  [VALIDATION] Validation failed for:
  python;/home/mas0047/fprime/fprime/cmake/support/validation/pipsetup.py
Call Stack (most recent call first):
  /home/mas0047/fprime/fprime/cmake/support/validation/Validation.cmake:52 (setup_validation_command)
  /home/mas0047/fprime/fprime/cmake/FPrime.cmake:23 (include)
  CMakeLists.txt:32 (include)`

5. Sample Application Error on Linux:

`make gen_make all
Generating templates
make[1]: Entering directory '/fprime/fprime/Autocoders/Python/src/fprime_ac/generators/templates/component'
cheetah-compile cpp.tmpl
make[1]: cheetah-compile: Command not found
make[1]: *** [cpp.py] Error 127
make[1]: Leaving directory '/fprime/fprime/Autocoders/Python/src/fprime_ac/generators/templates/component'
/fprime/fprime/mk/makefiles/templates.mk:3: recipe for target 'compile-templates' failed
make: [compile-templates] Error 2`

---

May I request feedback on any of the aforementioned errors to get the sample application running (Windows or Linux)?

Thanks!

LeStarch, 2019-07-22:
Currently, we do not support a Windows build.  Hence, the errors you get on Windows.

The error you see on CMake + Linux is reporting that the necessary PIP packages are not installed.  In order to make the build pass:

## If using Python 2
```
pip install -r mk/pip_required_build.txt 
```

## If using Python 3
```
pip install -r mk/pip_required_build3.py
```

ecruz480, 2019-07-27:
I had the same problem, the fix for me was installing cheetah
`sudo apt-get install python-cheetah`

msyed-aubbie, 2019-07-28:
> 
> 
> Currently, we do not support a Windows build. Hence, the errors you get on Windows.
> 
> The error you see on CMake + Linux is reporting that the necessary PIP packages are not installed. In order to make the build pass, install the necessary pip dependencies.
> ## If using Python 2
> 
> ```
> pip install -r mk/pip_required_build.txt 
> ```
> 
> ## If using Python 3
> 
> ```
> pip install -r mk/pip_required_build3.py
> ```

Many thanks and both solutions worked wonderfully depending on Python installation. One edit though:
The correct path is:
mk/python/pip_required_build.txt 
and
mk/python/pip_required_build3.py
This according to the latest Git Clone I have.
Thanks again!

LeStarch, 2019-09-03:
Ok, closing.

closed 2019-09-03 by LeStarch