nasa/fprime #60: Error when issuing 'make rebuild' in Math Component Tutorial

ecruz480, 2019-07-27:
When trying to build the `MathSender` component, I keep running into the same error when issuing the `make rebuild` command within the `Ref` directory. I have check the `mod.mk` file in the `configs/modules` folder and tried rearranging the module order under `REF_MODULES :=\`. Any help is appreciated.

[image]

timcanham, 2019-07-29:
It looks like there is something incorrect in the `MathSenderComponentAi.xml`. Check it against the reference copy in `docs/Tutorials/MathComponent/MathSender/MathSenderComponentAi.xml`.

ecruz480, 2019-07-31:
Hm, I must of missed something small, works fine now. Thank you!

closed 2019-07-31 by ecruz480