nasa/fprime #61: Adding Compile Commands to make system

brh68, 2019-08-01:
As part of our software, we are using a library called WiringPi, and in order to compile those files the compilier must be given the flag `-lwiringPi`.  How can we add this flag to the make system so that it will be able to successfully compile our drivers?
Ex:
[image]

timcanham, 2019-08-01:
You can add `-lwiringPi` to this line: https://github.com/nasa/fprime/blob/master/mk/configs/compiler/raspian_gnu_common.mk#L28

brh68, 2019-08-01:
I got is to that a `make rebuild` command will compile successfully, but the `make rebuild_rpi_cross` command gives a `no such file or directory #include <wiringPi.h>`, which is strange since the normal rebuild is able to find it.  Is there a certain place it needs to be installed, because I currently have WiringPi in my VM's home directory.

brh68, 2019-08-01:
I installed WiringPi in the same directory as the cross compile tools and it works, is this the correct way to do it?

timcanham, 2019-08-01:
That's one way, or you can put the wiringPi in `/opt` somewhere and add the include path here:

https://github.com/nasa/fprime/blob/master/mk/configs/compiler/raspian_gnu_common.mk#L45

It's better to keep the compile tools and wiringPi disentangled.

brh68, 2019-08-01:
I'm not sure how to add the path, Im not familiar with the syntax of makefiles

brh68, 2019-08-01:
[image]

brh68, 2019-08-04:
Im not sure how to correct the above error

timcanham, 2019-08-04:
Add a `-I` to the beginning of the `/opt/` path you've added.

brh68, 2019-08-05:
Earlier when I thought it worked, I actually had just ran `make rebuild`, so I haven't gotten `make rebuild_rpi_cross`.  After adding `-I`, it can't find the file that I specifically included.
[image]
[image]

timcanham, 2019-08-05:
Sorry, my bad: `-I/opt/tools/rpi/WiringPi/wiringPi` Then your include should work.

brh68, 2019-08-05:
That fixed the include problem, but now it is acting like wiringPi isn't installed, so I think the cross compiler can't tell it is installed on the VM, and I'm not sure how to correct this.
[image]

LeStarch, 2020-03-03:
I am closing this issue, as I see no updates.  Please feel free to continue discussion on our mailing list:  https://groups.google.com/forum/#!forum/fprime-community

Or open a new issue if you detect a new bug.

closed 2020-03-03 by LeStarch