nasa/fprime #62: Opcode encryption and verification

aaronbuchwald, 2019-08-01:
Hello,

Does the fprime communication framework provide any built-in encryption of commands sent from the ground station in order to ensure that no malicious attackers could send opcodes to our satellite? Additionally, are there any systems in place in order to prevent spurious commands from being accidentally read in by fprime and sent as an opcode?

Best,
A

timcanham, 2019-08-01:
It doesn't. We are just starting to think about it at JPL, and are exploring where it fits best. It is not uncommon for it to be supported at the radio level. FPrime does validate opcodes against a list of registered opcodes before invoking a component. The component also has verification of the correct data size for a command. Additionally, it is expected that if the command makes it through these two levels that the user would validate the argument values.

aaronbuchwald, 2019-08-01:
Awesome thanks Tim

closed 2019-08-01 by aaronbuchwald