nasa/fprime #63: How to make ut_raspian_cross work

aaronbuchwald, 2019-08-01:
Hello,

I am trying to make an I2C component for the raspberry pi and am not sure how to make/compile the unit test to go on the pi. I am following the directions here: https://github.com/fprime-community/fprime-course-materials/blob/master/Flight_Software_Workshop-June_2019/F-Prime-Command-Cheat-Sheet.md
However, this simply says that I need to add the two lines for ut_raspian_cross and ut_raspian_cross_clean, but this results in an exception that says: no rule to make target.

Best,
A

aaronbuchwald, 2019-08-01:
[image]

LeStarch, 2019-08-05:
After adding those commands, did you rerun :

```
make gen_make
```

aaronbuchwald, 2019-08-05:
```
# This Makefile goes in each module, and allows building of an individual module library.
# It is expected that each developer will add targets of their own for building and running
# tests, for example.

# derive module name from directory

MODULE_DIR = Drv/LinuxI2CDriver
MODULE = $(subst /,,$(MODULE_DIR))

BUILD_ROOT ?= $(subst /$(MODULE_DIR),,$(CURDIR))
export BUILD_ROOT

include $(BUILD_ROOT)/mk/makefiles/module_targets.mk

# Add module specific targets here

ut_raspian_cross:
	@$(TIME) $(MAKE) -f $(BUILD_ROOT)/mk/makefiles/Makefile COMP=gcc-cross BUILD=RASPIAN test_$(MODULE)testut

ut_raspian_cross_clean:
	@$(TIME) $(MAKE) -f $(BUILD_ROOT)/mk/makefiles/Makefile COMP=gcc-cross BUILD_RASPIAN test_$(MODULE)testut_clean
```

aaronbuchwald, 2019-08-05:
```

#
#   Copyright 2004-2008, by the California Institute of Technology.
#   ALL RIGHTS RESERVED. United States Government Sponsorship
#   acknowledged.
#
#

SRC = LinuxI2CDriverComponentAi.xml LinuxI2CDriverComponentImpl.cpp

HDR = LinuxI2CDriverComponentImpl.hpp

# SUBDIRS = test
```

aaronbuchwald, 2019-08-05:
I ran
```
make gen_make
```
but the output of
```
make ut_raspian_cross
```
is still the same: no rule to make target. Here is my makefile and mod.mk file. Thanks for the help.

timcanham, 2019-08-05:
You have `# SUBDIRS = test` still in there? That excludes the subdirectory to the unit tests.

aaronbuchwald, 2019-08-06:
Yes, I was working off of the RobotArm tutorial, but I was under the impression that that directory and its contents were autocoded by make ut_raspian_cross and then changed to edit the files.

aaronbuchwald, 2019-08-12:
What is make ut_raspian_cross supposed to do? Is there documentation anywhere? I thought that it would autocode the tests to be filled out as with other parts of FPrime.

timcanham, 2019-08-13:
There is a general pattern of writing unit tests that is covered in the tutorial. You generate the files using the `make testcomp` target and then copy them into the `test/ut` subdirectory. The standard way to compile the unit tests is to compile and run them on the host (e.g. Linux or Mac). The `make ut_raspian_cross` is a target I put in the local make files to allow you to cross-compile a unit test to run on the target itself - this is a way to test things like drivers that need the hardware.

closed 2019-08-30 by aaronbuchwald