nasa/fprime #64: Failed Add/Sub operation test after make run_ut

ecruz480, 2019-08-03:
After running `make run_ut`, I receive the following unit test results

[image]

I tried rerunning the code after copying the `MathSender` files from the Tutorial folder into my own directory but get the same failed results.

timcanham, 2019-08-04:
Hmmmm... Double-checked mine and still works okay... Do you have a branch?

ecruz480, 2019-08-05:
I have not set up a branch, I've only been grabbing files from the Tutorial folder when I was to cross check any errors I have received. I was reading through issue #22 and found many of the same problems. Kept saying `Permission denied` until I ran `chmod u+x runtest_LINUX` but followed up with the run test errors. I've looked at GTestBase.cpp and Tester.cpp but couldn't make out the problem prompted in the terminal. If there is some code fragment you believe could be the root cause I will post asap but I will brush back through the xml and C++ files I modified in `MathSender`.

ecruz480, 2019-08-05:
Tried repeating the error earlier today and got a different result saying I was missing:

`MathSenderComponentAc.cpp`
`GTestBase.cpp`
`TesterBase.cpp` (and each of their respective .hpp headers)

I merged the missing files from my old MathSender with the tutorial file and it all seemed to pass fine, not sure which bit of mine was causing the error.

closed 2019-08-05 by ecruz480