nasa/fprime #66: Sample Application make Command Error

msyed-aubbie, 2019-08-17:
Hello,

OS: Cygwin, all Python packages installed, make gen_make completed with no errors.

While running the 'make' command after 'make gen_make' for the sample application in the Ref directory, we encountered the following error:

mkdir -p /home/Saidah/fprime/Os/Posix/linux-cygwin-x86-debug-gnu-bin
/usr/bin/g++ -DASSERT_FILE_ID=0xC2E71506  -MMD -MP -MF/home/Saidah/fprime/Os/Posix/linux-cygwin-x86-debug-gnu-bin/Mutex.d -g3 -DTGT_OS_TYPE_LINUX -DISF -DBUILD_CYGWIN -Wall -Wextra -fno-builtin -fno-asm -fstrength-reduce -Wno-unused-parameter -Wno-long-long  -fcheck-new -Wnon-virtual-dtor  -m32 -I/home/Saidah/fprime/Fw/Types/Linux -I/home/Saidah/fprime -I/home/Saidah/fprime/gtest/include -I/home/Saidah/fprime/gtest -fno-rtti -I/home/Saidah/fprime/Os/Posix/  -c -o /home/Saidah/fprime/Os/Posix/linux-cygwin-x86-debug-gnu-bin/Mutex.o /home/Saidah/fprime/Os/Posix/Mutex.cpp
**mkdir -p /home/Saidah/fprime/Os/Linux/linux-cygwin-x86-debug-gnu-bin
/home/Saidah/fprime/Os/Posix/Mutex.cpp: In constructor ‘Os::Mutex::Mutex()’:
/home/Saidah/fprime/Os/Posix/Mutex.cpp:23:52: error: ‘PTHREAD_PRIO_INHERIT’ was not declared in this scope
         stat = pthread_mutexattr_setprotocol(&attr,PTHREAD_PRIO_INHERIT);
                                                    ^~~~~~~~~~~~~~~~~~~~
/home/Saidah/fprime/Os/Posix/Mutex.cpp:23:52: note: suggested alternative: ‘PTHREAD_ONCE_INIT’
         stat = pthread_mutexattr_setprotocol(&attr,PTHREAD_PRIO_INHERIT);
                                                    ^~~~~~~~~~~~~~~~~~~~
                                                    PTHREAD_ONCE_INIT
make[1]: *** [/home/Saidah/fprime/mk/makefiles/bin_targets.mk:5758: /home/Saidah/fprime/Os/Posix/linux-cygwin-x86-debug-gnu-bin/Mutex.o] Error 1
make[1]: *** Waiting for unfinished jobs....**
/usr/bin/g++ -DASSERT_FILE_ID=0x22ABA4FB  -MMD -MP -MF/home/Saidah/fprime/Os/Linux/linux-cygwin-x86-debug-gnu-bin/FileSystem.d -g3 -DTGT_OS_TYPE_LINUX -DISF -DBUILD_CYGWIN -Wall -Wextra -fno-builtin -fno-asm -fstrength-reduce -Wno-unused-parameter -Wno-long-long  -fcheck-new -Wnon-virtual-dtor  -m32 -I/home/Saidah/fprime/Fw/Types/Linux -I/home/Saidah/fprime -I/home/Saidah/fprime/gtest/include -I/home/Saidah/fprime/gtest -fno-rtti -I/home/Saidah/fprime/Os/Linux/  -c -o /home/Saidah/fprime/Os/Linux/linux-cygwin-x86-debug-gnu-bin/FileSystem.o /home/Saidah/fprime/Os/Linux/FileSystem.cpp
make[1]: Leaving directory '/home/Saidah/fprime/Ref'
Command exited with non-zero status 2

---

May I request your feedback on what steps would lead to a successful compilation avoiding this error?

Thanks!

timcanham, 2019-08-22:
@msyed-aubbie Sorry for the long delay. We haven't supported cygwin for a while. The OS classes are no longer compatible. We suggest that you switch to Ubuntu 16.04. If you have a Windows machine, you can do this with a virtual machine or WSL under Windows.

LeStarch, 2019-09-03:
Closing, no more support for Cigwin.

closed 2019-09-03 by LeStarch