nasa/fprime #68: GDS/GSE - Python version and required libraries on Ubuntu 16

msyed-aubbie, 2019-08-26:
Good morning,

Quick questions on Python version/ libraries to run gds/gse starting from the sample application:

- Which version of Python do you recommend on Ubuntu esp. for wxPython and Numpy/ Scipy?

(Issues: 

1. For v 2.7, a required library 'numpy' does not install saying it is geared to 3.5+. 
2. If 3.5/3.7+ are installed and used instead, several 'print' commands fail because those are geared to 2.7 with no parenthesis '( )'. Even after fixing files for parenthesis, Python 3.7 failed.

)

- Exactly which scripts need run prior to running the sample application or the Ground Data System in general? The options are in 'fprime/mk/python':

- pip_required_gds.txt
- pip_required_gse.txt
- pip_required_scripts.txt
- wxPython.sh

- Are all these aforementioned scripts need run prior to launching the sample application through: 'Ref/scripts/run_ref.sh' 

---

We have been able to make the application successfully but now there is a road block with the GDS starting from the sample application due to Python versioning and libraries. Would highly appreciate any feedback!

Thanks!

msyed-aubbie, 2019-09-03:
Many thanks!

I decided to try Debian and Python updates worked. FP framework now builds and GDS/GSE is displayed.

A follow up question:

The GUI for GDS/ GSE only stays on for 10 - 15 seconds and is closed after. I have tried several approaches but it just does not stay on longer.

May I please request any pointers on this issue?

Thanks!

Oxymoren, 2019-09-03:
I can't say that I've had that happen. Since my original comment, I've also set up an environment on Ubuntu 18.04. Have you run apt-get update/upgrade? Are there any errors that display when the GUI closes?

msyed-aubbie, 2019-09-03:
Thanks again!
Yes, initially the complaint was negative tab distance '15c' in line 118 of 'tkGui_misc.py':

`self.text.tag_config("hyper", font=font, foreground="blue", underline=1, justify=justify, tabs='15c')` 

We decided to comment the 'tabs' parameter out. No complaints now but the screen only stays on for 10 - 15 seconds and no errors shown when it closes. Will try Ubuntu 18.04 tonight.

msyed-aubbie, 2019-09-04:
Good morning and many thanks for the help!

Up and running now with the framework and the GDS/ GSE in Ubuntu 18.04 + Python 2.7.

closed 2019-09-04 by msyed-aubbie