nasa/fprime #69: Issue with using Fw::Buffer

aaronbuchwald, 2019-08-31:
I am getting some issues when using a U8 array with the Fw::Buffer. It says that it cannot convert from U8 to U64. I thought that Fw::Buffer would be a buffer of bytes and therefore should use U8, additionally this seems to be the way that @timcanham uses it in the RobotArm tutorial for the i2c component, so I assume there's a way to use it this way. Here's a screenshot of the error and my fork is at https://github.com/aaronbuchwald/fprime on the branch i2c-driver.

aaronbuchwald, 2019-08-31:
[image]

aaronbuchwald, 2019-08-31:
[image]

timcanham, 2019-09-02:
@aaronbuchwald Because there is a conversion from a pointer type (`U8*`) to the data member of `Fw::Buffer`, you need a cast of `data` to `U64`. Try:

```c++
dataBuffer.setdata(static_cast<U64>(data));
```

aaronbuchwald, 2019-09-03:
[image]

aaronbuchwald, 2019-09-10:
I got past this, but now there is a number of undefined reference errors in auto-generated files. Do you know what might cause this behavior?

aaronbuchwald, 2019-09-10:
[image]

aaronbuchwald, 2019-09-10:
[image]

aaronbuchwald, 2019-09-10:
Still working on the same branch as I referenced before

timcanham, 2019-09-11:
@aaronbuchwald Make sure that all the directories with the code you need are in the `LinuxI2CDriver/test/ut/mod.mk` file. You are probably missing some.

closed 2019-11-20 by aaronbuchwald