nasa/fprime #84: Ground GUI crashed after invalid commands

pelmini, 2019-09-22:
Submitted invalid command to ground station html gui, and the following crash happened: 
```
[2019-09-21 19:50:46,178] ERROR in app: Exception on /commands/cmdDisp.CMD_TEST_CMD_1 [PUT]
Traceback (most recent call last):
  File "/Users/fprime-venv/lib/python3.7/site-packages/flask/app.py", line 1949, in full_dispatch_request
    rv = self.dispatch_request()
  File "/Users/fprime-venv/lib/python3.7/site-packages/flask/app.py", line 1935, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/Users/fprime-venv/lib/python3.7/site-packages/flask_restful/__init__.py", line 458, in wrapper
    resp = resource(*args, **kwargs)
  File "/Users/fprime-venv/lib/python3.7/site-packages/flask/views.py", line 89, in view
    return self.dispatch_request(*args, **kwargs)
  File "/Users/fprime-venv/lib/python3.7/site-packages/flask_restful/__init__.py", line 573, in dispatch_request
    resp = meth(*args, **kwargs)
  File "/Users/Documents/fprime/Gds/src/fprime_gds/flask/commands.py", line 97, in put
    self.sender.send_command(cmd_name_dict[command].opcode, arg_list)
  File "/Users/Documents/fprime/Gds/src/fprime_gds/common/pipeline/standard.py", line 217, in send_command```
    cmd_data = fprime_gds.common.data_types.cmd_data.CmdData(tuple(args), command_template)
  File "/Users/Documents/fprime/Gds/src/fprime_gds/common/data_types/cmd_data.py", line 72, in __init__
    self.convert_arg_value(val, typ)
  File "/Users/Documents/fprime/Gds/src/fprime_gds/common/data_types/cmd_data.py", line 171, in convert_arg_value
    arg_type.val = int(arg_val)
ValueError: invalid literal for int() with base 10: 'x'

LeStarch, 2020-01-22:
I think this is fixed.  Need to verify.

LeStarch, 2020-03-03:
Ground system does not crash, but just reports the error to HTML.

LeStarch, 2020-04-02:
Full fix in-progress.

LeStarch, 2020-05-26:
Fix should be delivered.

closed 2020-05-26 by LeStarch