nasa/fprime #89: CMake Build Error

acxz, 2019-11-01:
Hello, I tried to build fprime, however I got the following error:

```
-- The C compiler identification is GNU 9.2.0
-- The CXX compiler identification is GNU 9.2.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- F´ core directory set to: /home/acxz/vcs/git/github/nasa/fprime/cmake/..
-- [VALIDATION] Running: python;/home/acxz/vcs/git/github/nasa/fprime/cmake/support/validation/pipsetup.py
enum34 is missing.
Markdown is missing.
Cheetah3; python_version > '2.7' is missing.
CMake Error at cmake/support/validation/Validation.cmake:38 (message):
  [VALIDATION] Validation failed for:
  python;/home/acxz/vcs/git/github/nasa/fprime/cmake/support/validation/pipsetup.py
Call Stack (most recent call first):
  cmake/support/validation/Validation.cmake:52 (setup_validation_command)
  cmake/FPrime.cmake:23 (include)
  CMakeLists.txt:11 (include)

-- Configuring incomplete, errors occurred!
```

I have python installed, but it seems that fprime's cmake is not finding it.

Steps to reproduce:
1. Clone fprime
2. cd fprime
3. mkdir build_dir
4. cd build_dir
5. cmake ..
6. See error above

System:
OS: ArchLinux
cmake: 3.15.5
python: 3.7.4

LeStarch, 2019-11-01:
This is a validation warning.  The following packages are not installed in the python you are using.
```
enum34 is missing.
Markdown is missing.
Cheetah3; python_version > '2.7' is missing.
```
If you installed F´ in a virtual environment, you must source the venv.  If not, you should install the packages in your system python.

acxz, 2019-11-01:
I checked those modules as well. enum34 comes in python (3.7.4) by default and I have markdown installed.

acxz, 2019-11-01:
The referenced PR fixes the issue for me. I believe the call to `python .../pipsetup.py` was not happening correctly since CMake adds semicolons between list arguments. (At least in the latest stable version 3.15.5) rather then spaces.

@LeStarch can you confirm if the the PR affects older versions of CMake (which I am assuming you are on, since you did not receive this cmake error)

acxz, 2019-11-01:
hmm maybe I might be misguided. So while the PR does result in cmake working properly, I didn't have cheetah3 installed. Here is my updated build log:

```
-- F´ core directory set to: /home/acxz/vcs/git/github/nasa/fprime/cmake/..
-- [VALIDATION] Running: python;/home/acxz/vcs/git/github/nasa/fprime/cmake/support/validation/pipsetup.py
enum34 is missing.
CMake Error at cmake/support/validation/Validation.cmake:38 (message):
  [VALIDATION] Validation failed for:
  python;/home/acxz/vcs/git/github/nasa/fprime/cmake/support/validation/pipsetup.py
Call Stack (most recent call first):
  cmake/support/validation/Validation.cmake:52 (setup_validation_command)
  cmake/FPrime.cmake:23 (include)
  CMakeLists.txt:11 (include)

-- Configuring incomplete, errors occurred!
See also "/home/acxz/vcs/git/github/nasa/fprime/build_dir/CMakeFiles/CMakeOutput.log".
```

System:
OS: ArchLinux
cmake: 3.15.5
python: 3.7.4
python-markdown: 3.1.1
python-cheetah3: 3.2.4
python-enum: Installed by defualt on python 3.7

acxz, 2019-11-01:
Okay I believe I have found the real issue and fixed, see #91 
I'll close this issue once the PR gets merged.

LeStarch, 2019-11-01:
This is a good fix.  Thanks for adding it!

PR #91 (2019-11-01): specifiy python version for enum34 dep
Since `enum` is part of python for newer versions, checking for `enum34` breaks the cmake process if you have a newer python version. This PR ensures `enum34` is checked for only if it pertains to the correct python version, specifically python <= 3.4.

Resolves #89
closed 2019-11-01 by LeStarch