nasa/fprime #93: Make Build Error

acxz, 2019-11-01:
I am trying to build fprime and I encountered the following issue:
```
[  0%] Generating ../../../../../../../../Autocoders/Python/src/fprime_ac/generators/templates/topology/includes1TopologyCpp.py, ../../../../../../../../Autocoders/Python/src/fprime_ac/generators/templates/topology/includes1TopologyH.py, ../../../../../../../../Autocoders/Python/src/fprime_ac/generators/templates/topology/publicInstanceTopologyCpp.py, ../../../../../../../../Autocoders/Python/src/fprime_ac/generators/templates/topology/publicInstanceTopologyH.py, ../../../../../../../../Autocoders/Python/src/fprime_ac/generators/templates/topology/publicTopologyCpp.py, ../../../../../../../../Autocoders/Python/src/fprime_ac/generators/templates/topology/publicTopologyH.py, ../../../../../../../../Autocoders/Python/src/fprime_ac/generators/templates/topology/publicTopologyID.py
/bin/sh: cheetah-compile: command not found
make[2]: *** [F-Prime/Autocoders/Python/src/fprime_ac/generators/templates/CMakeFiles/codegen.dir/build.make:276: ../Autocoders/Python/src/fprime_ac/generators/templates/topology/includes1TopologyCpp.py] Error 127
make[1]: *** [CMakeFiles/Makefile2:309: F-Prime/Autocoders/Python/src/fprime_ac/generators/templates/CMakeFiles/codegen.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```

Steps to reproduce:

1. Clone fprime
2. cd fprime
3. mkdir build_dir
4. cd build_dir
5. cmake ..
6. make
7. See error above

System:
OS: ArchLinux
python-cheetah3: 3.2.4

I believe the issue is caused because `python-cheetah3` ships with `cheetah-compile3` instead of `cheetah-compile` as in `python2-cheetah` in ArchLinux. I checked the Debian package and it is the same thing, Debian `python3-cheetah` ships with `cheetah-compile3` instead of `cheetah-compile`. (https://packages.debian.org/de/sid/amd64/python3-cheetah/filelist) This makes me believe that `fprime` does not build under the latest python (v3) at all. I can change all the `cheetah-compile`s to `cheetah-compile3`s but that would make it incompatible for older pythons. Maybe we can add a check every time we need to call `cheetah-compile` to make sure the correct one is being used? I am not sure about a good solution for this issue.

LeStarch, 2019-11-01:
If cheetah3 is installed via pip, it comes with the `cheetah-compile` utility. It appears the packaging authorities have renamed the utility for system packages.

Typically we use pip to install the Python dependencies, and I would expect that would resolve the issue.

acxz, 2019-11-01:
I see, I would like to try to get this to build system-wide instead of using a virtualenv. I want to see if I can come up with a clean solution to this issue.

acxz, 2019-11-02:
@LeStarch I think I have found a solution, I tried to make it as elegant as possible. This was a hard one for me, but it works.

LeStarch, 2019-11-04:
I am still confused why this is an issue.  A user can install python packages using pip into the system python (See Debian-like instructions below).  So why try to support generic system packages?

## Debian-like No-Venv Instructions
```
apt install python-pip
pip install -r fprime/mk/python/....
```

I am hesitant to try and support generic system packages, because there is no guarantee that the system packaged versions work.

In the requirements.txt files, and pip packages, specific versions of dependencies can be set. With system packages we don't have control over dependency versions, and thus cannot ensure that they work together.  When installed via PIP, the specific working-set can be specified.

@timcanham what do you think on this issue?

timcanham, 2019-11-04:
What were the issues installing the packages to the system python? I've done `sudo pip install XXXX` many times.

acxz, 2019-11-05:
Here is a description for why Python does not recommend installing pip packages system wide: https://docs.python.org/3/installing/index.html#installing-into-the-system-python-on-linux

Maybe it is not an issue, if `fprime`'s requirements are to only work in `venv`. While it is true installing over `pip` gives `fprime` more control over its dependencies, I believe being able to use system wide dependencies is not too hard to incorporate and I am sure some people will appreciate it, at the very least I will and I am willing to spend the time to get this resolved.

PR #94 (2019-11-18): Cheetah compile py3 fix
This allows template code generation to occur using a system installation of `python-cheetah3`. 
We check if we are using python 2 or if we not using a system install of python, i.e. if we are using a venv. If that is satisfied then we use `cheetah-compile`. Else if we are using a system install of python3 then we use `cheetah-compile3`. This fixes build issues on Arch and Debian based systems.

One caveat is that if you are building `fprime` with python 3 outside a venv/virtualenv then you cannot use `pip`'s version of `cheetah3`. This is fine though, since using pip a [... cut]
closed 2019-11-18 by LeStarch