nasa/fprime #100: run_ref_gds.sh script is outdated

matzipan, 2019-12-02:
Following the readme at https://github.com/nasa/fprime/blob/master/Gds/README.md#gds-gui-usage, I get the following error:

```
$ ./Ref/scripts/run_ref_gds.sh 
usage: run_deployment.py [-h] [-d DEPLOY] [-l LOGS] [--log-directly]
                         [--tts-port TTS_PORT] [--tts-addr TTS_ADDR]
                         [-g {none,html}] [--dictionary DICTIONARY]
                         [-c CONFIG] [--comm-adapter {ip}] [-n] [--app APP]
run_deployment.py: error: argument -g/--gui: invalid choice: 'wx' (choose from 'none', 'html')
```

I'm assuming this is due to the recent change to the HTML gds. 

I changed `-g` to `html` in the script and now I get the following: 
```
[ERROR] Could not find python dictionary no XML dictionary in /home/andrei/Workspace/fprime/Ref
```
The documentation does not mention any dictionary, so either the script or the documentation is wrong. 

I would gladly submit a PR as soon as I'm pointed to what needs to change.

matzipan, 2019-12-02:
I have edited the original report to reflect that the second error was due to me not having built the reference application.

LeStarch, 2019-12-02:
Ok, yes.  Once you build Ref and run `make install` then all you need to do is choose `html` as the GUI.   You can still use wx, but the installation is often complicated, so we removed it as a default from the pacakge.

LeStarch, 2019-12-02:
Honestly, we probably need to update that whole readme section.  I'll look into that to.

closed 2020-03-03 by LeStarch