nasa/fprime #102: gen_make for reference application does not build successfully

matzipan, 2019-12-02:
Hi, 

In order to be able to build the make files for the reference applications I had to apply the following patch:
```
diff --git a/mk/configs/modules/modules.mk b/mk/configs/modules/modules.mk
index 0a7b67ee..454b01cd 100644
--- a/mk/configs/modules/modules.mk
+++ b/mk/configs/modules/modules.mk
@@ -45,15 +45,11 @@ SVC_MODULES := \
        Svc/CmdDispatcher \
        Svc/CmdSequencer \
        Svc/Seq \
-       Svc/GndIf \
        Svc/ActiveRateGroup \
        Svc/RateGroupDriver \
        Svc/Sched \
        Svc/ComLogger \
-       Svc/SocketGndIf \
-       Svc/BuffGndSockIf \
        Svc/TlmChan \
-       Svc/PassiveTextLogger \
        Svc/PassiveConsoleTextLogger \
        Svc/Time \
        Svc/Cycle \
```

Otherwise, `make gen_make` would exit with the following message:
```
**Error: File Svc/GndIf/mod.mk does not exist.**
Command exited with non-zero status 255
Build Time: 0:00.03
make: *** [/home/andrei/Workspace/fprime/mk/makefiles/deployment_makefile.mk:107: gen_make] Error 255
```

LeStarch, 2019-12-12:
@matzipan it looks like the older make system has diverged from the CMake one.  Your patch is correct, but we should add in:

```
   Svc/GroundInterface
```

Still looking into the compile errors.

david-kooi, 2019-12-19:
Having a similar issue with the RPI demo.

`compilation terminated.
/home/david/Workspace/fprime/mk/makefiles/bin_targets.mk:12848: recipe for target '/home/david/Workspace/fprime/RPI/Top/raspian-raspian-arm-debug-gnu-bin/Topology.o' failed
make[1]: *** [/home/david/Workspace/fprime/RPI/Top/raspian-raspian-arm-debug-gnu-bin/Topology.o] Error 1
In file included from /home/david/Workspace/fprime/RPI/Top/Components.hpp:9:0,
                 from /home/david/Workspace/fprime/RPI/Top/RPITopologyAppAc.cpp:2:
/home/david/Workspace/fprime/Svc/PassiveConsoleTextLogger/ConsoleTextLoggerImpl.hpp:4:73: fatal error: Svc/PassiveConsoleTextLogger/PassiveTextLoggerComponentAc.hpp: No such file or directory
 #include <Svc/PassiveConsoleTextLogger/PassiveTextLoggerComponentAc.hpp>
                                                                         ^
compilation terminated.
/home/david/Workspace/fprime/mk/makefiles/bin_targets.mk:12853: recipe for target '/home/david/Workspace/fprime/RPI/Top/raspian-raspian-arm-debug-gnu-bin/RPITopologyAppAc.o' failed
make[1]: *** [/home/david/Workspace/fprime/RPI/Top/raspian-raspian-arm-debug-gnu-bin/RPITopologyAppAc.o] Error 1
make[1]: Leaving directory '/home/david/Workspace/fprime/RPI'
Command exited with non-zero status 2
Build Time: 0:00.57
Makefile:20: recipe for target 'rpi_cross' failed
make: *** [rpi_cross] Error 2
`

LeStarch, 2020-03-03:
@timcanham is this fixed?

LeStarch, 2020-03-03:
Fixed by @timcanham , thanks!

closed 2020-03-03 by LeStarch