nasa/fprime #103: Building reference application with CMake doesn't get past generating the application topology

matzipan, 2019-12-02:
```
andrei@PC-40:~/Workspace/fprime/build$ cmake ../Ref/
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- F prime core directory set to: /home/andrei/Workspace/fprime/cmake/..
-- Target build platform: Linux
-- Including /home/andrei/Workspace/fprime/cmake/platform/Linux.cmake
-- Requiring thread library
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- [VALIDATION] Running: python;/home/andrei/Workspace/fprime/cmake/support/validation/pipsetup.py
-- F prime BUILD_ROOT currently set to: /home/andrei/Workspace/fprime/cmake/..
-- Installation directory: /home/andrei/Workspace/fprime/Ref
-- Adding library: Fw_Buffer
-- 	Found serializable: Buffer from BufferSerializableAi.xml
-- 	Found port: BufferGet from BufferGetPortAi.xml
-- 	Found port: BufferSend from BufferSendPortAi.xml
-- Adding library: Fw_Com
-- 	Found port: Com from ComPortAi.xml
-- Adding library: Fw_Cmd
-- 	Found port: Cmd from CmdPortAi.xml
-- 	Found port: CmdReg from CmdRegPortAi.xml
-- 	Found port: CmdResponse from CmdResponsePortAi.xml
-- Adding library: Fw_Log
-- 	Found port: Log from LogPortAi.xml
-- 	Found port: LogText from LogTextPortAi.xml
-- Adding library: Fw_Logger
-- Adding library: Fw_Time
-- 	Found port: Time from TimePortAi.xml
-- Adding library: Fw_Tlm
-- 	Found port: Tlm from TlmPortAi.xml
-- Adding library: Fw_Prm
-- 	Found port: PrmGet from PrmGetPortAi.xml
-- 	Found port: PrmSet from PrmSetPortAi.xml
-- Adding library: Fw_Cfg
-- Adding library: Fw_ComFile
-- Adding library: Fw_Comp
-- Adding library: Fw_Obj
-- Adding library: Fw_Port
-- Adding library: Fw_Types
-- Adding library: Fw_FilePacket
-- Adding library: Fw_SerializableFile
-- Adding library: Svc_Cycle
-- 	Found port: Cycle from CyclePortAi.xml
-- Adding library: Svc_Fatal
-- 	Found port: FatalEvent from FatalEventPortAi.xml
-- Adding library: Svc_Ping
-- 	Found port: Ping from PingPortAi.xml
-- Adding library: Svc_PolyIf
-- 	Found port: Poly from PolyPortAi.xml
-- Adding library: Svc_Sched
-- 	Found port: Sched from SchedPortAi.xml
-- Adding library: Svc_Seq
-- 	Found port: CmdSeqIn from CmdSeqInPortAi.xml
-- Adding library: Svc_WatchDog
-- 	Found port: WatchDog from WatchDogPortAi.xml
-- Adding library: Svc_ActiveLogger
-- 	Found component: ActiveLogger from ActiveLoggerComponentAi.xml
-- Adding library: Svc_ActiveRateGroup
-- 	Found component: ActiveRateGroup from ActiveRateGroupComponentAi.xml
-- Adding library: Svc_AssertFatalAdapter
-- 	Found component: AssertFatalAdapter from AssertFatalAdapterComponentAi.xml
-- Adding library: Svc_BufferManager
-- 	Found component: BufferManager from BufferManagerComponentAi.xml
-- Adding library: Svc_ComLogger
-- 	Found component: ComLogger from ComLoggerComponentAi.xml
-- Adding library: Svc_CmdDispatcher
-- 	Found component: CommandDispatcher from CommandDispatcherComponentAi.xml
-- Adding library: Svc_CmdSequencer
-- 	Found component: CmdSequencer from CmdSequencerComponentAi.xml
-- Adding library: Svc_FatalHandler
-- 	Found component: FatalHandler from FatalHandlerComponentAi.xml
-- Adding library: Svc_FileDownlink
-- 	Found component: FileDownlink from FileDownlinkComponentAi.xml
-- Adding library: Svc_FileManager
-- 	Found component: FileManager from FileManagerComponentAi.xml
-- Adding library: Svc_FileUplink
-- 	Found component: FileUplink from FileUplinkComponentAi.xml
-- Adding library: Svc_GroundInterface
-- 	Found component: GroundInterface from GroundInterfaceComponentAi.xml
-- Adding library: Svc_Health
-- 	Found component: Health from HealthComponentAi.xml
-- Adding library: Svc_LinuxTime
-- Adding library: Svc_LinuxTimer
-- 	Found component: LinuxTimer from LinuxTimerComponentAi.xml
-- Adding library: Svc_PassiveConsoleTextLogger
-- 	Found component: PassiveTextLogger from PassiveTextLoggerComponentAi.xml
-- Adding library: Svc_PolyDb
-- 	Found component: PolyDb from PolyDbComponentAi.xml
-- Adding library: Svc_PrmDb
-- 	Found component: PrmDb from PrmDbComponentAi.xml
-- Adding library: Svc_RateGroupDriver
-- 	Found component: RateGroupDriver from RateGroupDriverComponentAi.xml
-- Adding library: Svc_Time
-- 	Found component: Time from TimeComponentAi.xml
-- Adding library: Svc_TlmChan
-- 	Found component: TlmChan from TlmChanComponentAi.xml
-- Adding library: Os
-- Adding library: Drv_DataTypes
-- 	Found port: DataBuffer from DataBufferPortAi.xml
-- Adding library: Drv_GpioDriverPorts
-- 	Found port: GpioWrite from GpioWritePortAi.xml
-- 	Found port: GpioRead from GpioReadPortAi.xml
-- Adding library: Drv_SerialDriverPorts
-- 	Found port: SerialRead from SerialReadPortAi.xml
-- 	Found port: SerialWrite from SerialWritePortAi.xml
-- Adding library: Drv_SpiDriverPorts
-- 	Found port: SpiReadWrite from SpiReadWritePortAi.xml
-- Adding library: Drv_BlockDriver
-- 	Found component: BlockDriver from BlockDriverComponentAi.xml
-- Adding library: Drv_LinuxGpioDriver
-- 	Found component: LinuxGpioDriver from LinuxGpioDriverComponentAi.xml
-- Adding library: Drv_LinuxSerialDriver
-- 	Found component: LinuxSerialDriver from LinuxSerialDriverComponentAi.xml
-- Adding library: Drv_LinuxSpiDriver
-- 	Found component: LinuxSpiDriver from LinuxSpiDriverComponentAi.xml
-- Adding library: Drv_SocketIpDriver
-- 	Found component: SocketIpDriver from SocketIpDriverComponentAi.xml
-- Adding library: CFDP_Checksum
-- Adding library: Utils_Hash
-- Adding library: Utils_Types
-- Adding library: Ref_PingReceiver
-- 	Found component: PingReceiver from PingReceiverComponentAi.xml
-- Adding library: Ref_RecvBuffApp
-- 	Found serializable: PacketStat from PacketStatSerializableAi.xml
-- 	Found component: RecvBuff from RecvBuffComponentAi.xml
-- Adding library: Ref_SendBuffApp
-- 	Found component: SendBuff from SendBuffComponentAi.xml
-- Adding library: Ref_SignalGen
-- 	Found component: SignalGen from SignalGenComponentAi.xml
-- Adding executable: Ref
-- 	Found topology: Ref from RefTopologyAppAi.xml
-- Configuring done
-- Generating done
-- Build files have been written to: /home/andrei/Workspace/fprime/build
andrei@PC-40:~/Workspace/fprime/build$ make
Scanning dependencies of target codegen
[  0%] Building C object F-Prime/Autocoders/Python/src/fprime_ac/generators/templates/CMakeFiles/codegen.dir/home/andrei/Workspace/fprime/cmake/support/src/empty.c.o
[  1%] Linking C static library ../../../../../../../lib/Linux/libcodegen.a
[  5%] Built target codegen
Scanning dependencies of target RefTopologyAppAi_Ref_dict
[  5%] Generating RefTopologyAppAc.cpp, RefTopologyAppAc.hpp
Traceback (most recent call last):
  File "/home/andrei/Workspace/fprime/cmake/../Autocoders/Python/bin/codegen.py", line 1451, in <module>
    main()
  File "/home/andrei/Workspace/fprime/cmake/../Autocoders/Python/bin/codegen.py", line 1418, in main
    the_parsed_topology_xml = XmlTopologyParser.XmlTopologyParser(xml_filename)
  File "/home/andrei/Workspace/fprime/Autocoders/Python/src/fprime_ac/parsers/XmlTopologyParser.py", line 89, in __init__
    self.validate_xml(xml_file, element_tree, 'schematron', 'top_unique')
  File "/home/andrei/Workspace/fprime/Autocoders/Python/src/fprime_ac/parsers/XmlTopologyParser.py", line 247, in validate_xml
    if not validator_type in self.Config or not validator_name in self.Config[validator_type]:
TypeError: argument of type 'instance' is not iterable
make[2]: *** [Ref/Top/CMakeFiles/RefTopologyAppAi_Ref_dict.dir/build.make:62: Ref/Top/RefTopologyAppAc.cpp] Error 1
make[1]: *** [CMakeFiles/Makefile2:6550: Ref/Top/CMakeFiles/RefTopologyAppAi_Ref_dict.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

LeStarch, 2019-12-02:
Looks like an error using Python 2.  For now the recommendation is to use Python 3, however; I'll look into a Python 2 hotfix.

matzipan, 2019-12-02:
Interesting. Would you accept a pull request adding a CMake check for python3?

matzipan, 2019-12-02:
https://legacy.python.org/dev/peps/pep-0394/ states:
> The more general python command should be installed whenever any version of Python 2 is installed and should invoke the same version of Python as the python2 command (however, note that some distributions have already chosen to have python implement the python3 command; see the Rationale and Migration Notes below).

So in my very default Ubuntu setup, python calls python2 as per the PEP. Perhaps the call should be just changed to `python3` to remove any ambiguities?

LeStarch, 2019-12-02:
We could have CMake run a search-for-executable that looks for Python3 and then defaults to Python.  This would be good.  We have not officially ended our support for Python 2 yet, however; we will transition away soon due to Python 2's end-of-life in less than 1 month.

LeStarch, 2019-12-12:
I looked into replicating this error in Python 2, and couldn't do so. Did you install the dependencies?

CDKnightNASA, 2020-02-10:
I'm seeing this issue on my Ubuntu 19.10 x64 Desktop system. I've tried aliasing python to python3 but no joy. Where should I be looking for further logging information?

LeStarch, 2020-02-10:
Are you running inside a virtual environment or outside of a virtual environment?  If you run `python` on the command line, what version do you get?

CDKnightNASA, 2020-02-10:
Inside VMWare Workstation, typing the "python" command gives me 3.7.5. I'm going to try with Debian distros.

LeStarch, 2020-03-03:
Closing, we now offically depend on Python 3.5+ as this is the current non-end-of-life version of python.

closed 2020-03-03 by LeStarch