nasa/fprime #104: Add getting started scripts to the CI

matzipan, 2019-12-02:
Hi,

Newcomer here. We've been looking at fprime for a few weeks now and, personally, I've grown more and more excited to see that some of the ideas I had in mind for an on-board software suite you have already implemented much better than I would.

This week I managed to secure some time to play around with your framework. To my surprise, both the genmake and the CMake flow are not working for the reference deployment.

I'm confused whether this means that my setup is wrong or the flows are broken. I can't tell if CI fully covers them, as the CI scripts seem rather complex:

- `./run_all_unit_tests.sh` and `./run_ac_tests.sh` are part of ptf while `./mk/ci/cmake-based-ci.bash` is part of `mk/ci`.
- there is no single entry-point to the CI, there is a good deal of scripting inside the Jenkins config.xml. Perhaps if there is a single entry-point, that could also be used by developers and can also be used with travis-ci here on github? 
- `ptf/README` is 3 years old and has a note `This is in the process of being updated, so many features are still nonworking`. I am guessing this is no longer the case but unclear.

I'm willing to help with any of these tasks as long as I know my merge request stands a chance of being merged.

Regards,
Andrei

LeStarch, 2019-12-12:
Currently, GitHub is the best place to interact with us. We do appreciate PRs and try our best to merge them once we have discussed in the PR the implications to the code.

I agree with your assessments that the CI should have a single entry point.  As we move to a CMake based system, that single entry point is intended to be: https://github.com/nasa/fprime/blob/master/mk/ci/cmake-based-ci.bash and the PTF stuff will slowly fall away.

LeStarch, 2020-03-03:
Working on a fix for this.  It will be basically one entry point that distributes underneath.  We now also have a mailing list: https://groups.google.com/forum/#!forum/fprime-community

LeStarch, 2020-05-26:
Closing this issue.

closed 2020-05-26 by LeStarch