nasa/fprime #109: MagicDraw reports missing resources

matzipan, 2019-12-04:
Similar to issue #102, MagicDraw reports missing files:
```
The following resources are needed for this project, but missing in your MagicDraw installation. We recommend that you Upgrade your installation, installing missing resources.

Missing resources:
BuffGndSockIf.mdxml (missing)
PassiveTextLoggerModule.mdxml (missing)
GndIfModule.mdxml (missing)
SSCAEProjectUsageIntegrityProfile.mdzip (missing)
```

Also note the extra presence of `SSCAEProjectUsageIntegrityProfile.mdzip`. Is this file restricted?

LeStarch, 2019-12-04:
The following is typically ignored, as we don't have that file either:
```
SSCAEProjectUsageIntegrityProfile.mdzip (missing)
```

The following need cleaning up:
```
BuffGndSockIf.mdxml (missing)
PassiveTextLoggerModule.mdxml (missing)
GndIfModule.mdxml (missing)
```

LeStarch, 2020-04-02:
Fix in progress.

LeStarch, 2020-05-26:
Should be delivered.

closed 2020-05-26 by LeStarch