nasa/fprime #114: CosmosTopParser.py throws UnboundLocalError

matzipan, 2019-12-05:
```
$ ./run_cosmosgen.sh ~/Workspace/fprime/Ref/Top/RefTopologyAppAi.xml 
BUILD_ROOT is: /home/andrei/Workspace/fprime
make: *** No rule to make target 'print_python_base'.  Stop.
PYTHON_BASE: 
PYTHONPATH: /home/andrei/Workspace/fprime/Autocoders/Python/src
COSMOS_PATH: /home/andrei/Workspace/fprime/COSMOS

Using XmlParser and XmlTopologyParser instances
Parsing Component FatalHandler
Parsing Component ActiveRateGroup
Parsing Component PingReceiver
Parsing Component TlmChan
Parsing Component SendBuff
Parsing Component RecvBuff
Parsing Component CommandDispatcher
Parsing Component PrmDb
Parsing Component ActiveRateGroup
Parsing Component CmdSequencer
Parsing Component SignalGen
Parsing Component SignalGen
Parsing Component SignalGen
Parsing Component ActiveRateGroup
Parsing Component FileUplink
Parsing Component SignalGen
Parsing Component BufferManager
Parsing Component SignalGen
Parsing Component AssertFatalAdapter
Parsing Component Health
Parsing Component BufferManager
Parsing Component ActiveLogger
Parsing Component Time
Parsing Component RateGroupDriver
Parsing Component BlockDriver
Parsing Component FileDownlink
Parsing Component PassiveTextLogger
Parsing Component GroundInterface
Parsing Component LinuxTimer
Parsing Component SocketIpDriver
Finished Reusing XmlParser and XmlTopologyParser instances

Parsing Topology
Found 29 components.

Traceback (most recent call last):
  File "/home/andrei/Workspace/fprime/Autocoders/Python/bin/cosmosgen.py", line 187, in <module>
    main()
  File "/home/andrei/Workspace/fprime/Autocoders/Python/bin/cosmosgen.py", line 167, in main
    cosmos_parser.parse_topology(xml_filename)
  File "/home/andrei/Workspace/fprime/Autocoders/Python/src/fprime_ac/utils/cosmos/CosmosTopParser.py", line 164, in parse_topology
    cosmos_cmd.add_item(n, t, c, enum)
UnboundLocalError: local variable 'enum' referenced before assignment
```

SterlingPeet, 2020-04-01:
@LeStarch, is this one supposed to be closed?

LeStarch, 2020-04-01:
No, I think this issue is still outstanding for the cosmos stuff.

LeStarch, 2020-05-26:
Ported to fprime-cosmos

closed 2020-05-26 by LeStarch