nasa/fprime #132: Error in directions in INSTALL.md

aaronbuchwald, 2020-01-26:
https://github.com/nasa/fprime/blob/master/docs/INSTALL.md
After completing all of the instructions on this page, the final command to see if everything was successfully installed seems to be failing. It seems that the command in the documentation is not actually one of the options to use for fprime-util. I am trying to run:
```fprime-util -g html -d <path/to/fprime/checkout>/Ref```

However, when I run this I see the following result
[image]

LeStarch, 2020-01-27:
@aaronbuchwald good catch.  Yes, the GDS is a separate command.  I have updated the documentation for master.  Thanks!

aaronbuchwald, 2020-01-30:
Thanks for the quick fix!

closed 2020-01-30 by aaronbuchwald