nasa/fprime #134: Generating IMPL files Error

EbenezerA99, 2020-01-30:
I am trying to add a new component into the Ref folder. I have written my Ai.xml file but I am not able to generate the impl files. It seems that when I try to add a component into the Ref folder, I have to add lines in RefTopologyAppAI.xml to import my component type and an instance of my component in order for the 'fprime-util generate' to work. After I run the generate command I am able to go into my component folder and run the 'fprime-util impl' command to create the implementation files.  If I do not add the lines into the RefTopologyAppAI.xml and try to run 'fprime-util generate' in the Ref folder, it immediately errors. 

Here's my github fork:
https://github.com/EbenezerA99/fprime/tree/EPS

LeStarch, 2020-01-30:
Thanks.  I'll try to investigate.

aaronbuchwald, 2020-03-04:
Any news on this? I am running into the same issue now

LeStarch, 2020-03-06:
What is the error?  If you get the below error, you should do a build of ref first.

```
Traceback (most recent call last):
  File "...fprime/cmake/../Autocoders/Python/bin/implgen.py", line 46, in <module>
    from fprime_ac.generators.writers import ImplCppWriter
  File ".../fprime/Autocoders/Python/src/fprime_ac/generators/writers/ImplCppWriter.py", line 17, in <module>
    from fprime_ac.generators.templates.impl import cpp
ImportError: cannot import name 'cpp' from 'fprime_ac.generators.templates.impl' (.../fprime/Autocoders/Python/src/fprime_ac/generators/templates/impl/__init__.py)
```

```
cd Ref
fprime-util build
```

SterlingPeet, 2020-04-01:
Is this a case where better error messages would help?  If so, is a fix in the works?

LeStarch, 2020-04-01:
Some better errors are in the works, but I don't think we made this one better.  We'll do that.

LeStarch, 2020-05-26:
I think this is fixed in 1.4.

closed 2020-05-26 by LeStarch