nasa/fprime #135: Segmentation fault when running RPI demo

aaronbuchwald, 2020-01-30:
After completing the RPI demo step by step: https://github.com/nasa/fprime/blob/master/RPI/README.md

I get an immediate segmentation fault error when running the RPI binary on the raspberry pi. Ground station is running successfully when I do this and does not register any messages/telemetry from the RPI. Here is the output:

[image]

LeStarch, 2020-01-30:
@aaronbuchwald have you properly setup the Rapberry PI configuration?  This segmentation fault usually occurs when the GPIO has not been enabled.  Make sure your settings look like the settings in the "Board Preparation" part of that guide.

aaronbuchwald, 2020-01-30:
Ok I have enabled every interface in the Raspberry Pi configuration panel now. Also do you mean that Remote GPIO should be enabled? Because in the picture in the documentation it is shown as disabled. I have also made sure to follow the instructions for disabling UART.

I am still getting the exact same segmentation fault error.

timcanham, 2020-01-30:
Are you running as sudo? Sometimes as the regular user it won't let you open the device?

aaronbuchwald, 2020-01-31:
I'll try with sudo tomorrow.

Also, should Remote GPIO be disabled as in the image in the documentation or enabled?

timcanham, 2020-01-31:
Disabled. It has some other purpose we don't use.

aaronbuchwald, 2020-01-31:
Got rid of the segmentation fault, but it is still not able to connect to the ground station server. On the raspberry pi it simply prints out hit ctrl c to quit and then does nothing. The ground station does not react at all and there is a red x in the top right, which I assumed indicates it's not connected to anything. I tried adding the parameters
```
-a 127.0.0.1 -p 50000
```
to the script running on the ground station, but this didn't change anything.

I went through logs for the last run and the interesting thing is the comm-adapter log:

[image]

aaronbuchwald, 2020-01-31:
How can I get it to connect? I made sure that they are both running on the same WiFi network already and not sure what else to try.

timcanham, 2020-01-31:
`127.0.0.0` is a private address, so the RPI can't connect to it there. What is the command line on the RPI side?

aaronbuchwald, 2020-02-01:
On RPI side it is:
```
sudo ./RPI -a 10.0.2.15 -p 50000
```
And I also tried it on the laptop side without those parameters added (as in the documentation I was just playing around and trying to get it to work), which produced the same result.

LeStarch, 2020-02-01:
According to your logs, you are connecting the GDS server to 0.0.0.0, which is correct.  That says "listen to all incoming connections".

Usually this issue happens when the firewall on the ground system is fighting you.  Try the following:

```
ufw allow 50000
```

aaronbuchwald, 2020-02-03:
Ran this and then enabled the firewall, but it did not have any effect. I realize now I should have mentioned earlier, that I'm running the ground station from within virtualBox, so this is most likely a port forwarding issue.

timcanham, 2020-02-03:
That could be it. In the past, I've used a USB network adapter and attached it to the VM so that the guest OS can set the network address, firewall, etc.

LeStarch, 2020-02-04:
You need to forward port 50000 from the host into virtual box.  Typically I
forward it with the server address "0.0.0.0", at least to get going.

LeStarch, 2020-02-17:
@aaronbuchwald did this work?  Can we close this?

aaronbuchwald, 2020-02-17:
No, this didn't work for me. Figuring out a different solution now. Topic has shifted from original post, so might as well close

closed 2020-02-17 by aaronbuchwald