nasa/fprime #137: Adding WiringPi to the FPrime repository

scbrechter, 2020-02-09:
My name is Stefan Brechter and I'm with the CisLunar Explorers cubesat team at Cornell University. My team and I have found that using WiringPi has made it much easier for our fprime components and drivers to interface with hardware and sensors on the cubesat. We are running into an issue when we try compiling components that include the wiringpi library. We're not sure what file needs to be changed to correctly include WiringPi so that fprime sees it and knows it is there.

Do you have any idea about what files need to be edited in order to resolve this issue? Could it be a good idea to include WiringPi in the default FPrime repo?
[image]

LeStarch, 2020-02-09:
Are you compiling with CMake and "fprime-util" or "make"?

scbrechter, 2020-02-16:
When we ran into this issue we were still compiling with "make". Our team is going through the GPS tutorial and shifting our work to the updated CMake version of F Prime now.

LeStarch, 2020-05-26:
Did we close the loop on this effort?

LeStarch, 2020-08-03:
Closing.  Reopen if you still have issues with CMake.

closed 2020-08-03 by LeStarch