nasa/fprime #141: Remove unnecessary null pointer checks

elfring, 2020-02-15:
[An extra null pointer check is not needed in functions](https://isocpp.org/wiki/faq/freestore-mgmt#delete-handles-null "Do I need to check for null before delete p?") like the following.
- [BufferQueue::finalize](https://github.com/nasa/fprime/blob/44bc57574daaac8d455b60912f664e6c57d487d9/Os/Pthreads/PriorityBufferQueue.cpp#L101 "BufferQueue::finalize function")
- [RandomlyBoundedScenario](https://github.com/nasa/fprime/blob/c35ab7ee882e2de906e731d525573b6aba560c9e/STest/STest/Scenario/RandomlyBoundedScenario.hpp#L48 "Destructor for RandomlyBoundedScenario")

timcanham, 2020-02-15:
Good point! Make a fix and do a PR and we'll incorporate it.

SterlingPeet, 2020-04-01:
@timcanham : can we actually guarantee this is the case?  Sometimes custom ``new`` and ``delete `` implementations are needed for smaller platforms, e.g.: like 8-bit AVRs.  Then again, those implementations could include the protection there...

SterlingPeet, 2021-03-03:
This issue is stale, and can likely be closed.

LeStarch, 2021-03-04:
Agreed.  It can be reopened if needed.

closed 2021-03-04 by LeStarch