nasa/fprime #142: Improve exception safety with smart pointers

elfring, 2020-02-15:
Would you like to [wrap any pointer data members](https://github.com/nasa/fprime/blob/44bc57574daaac8d455b60912f664e6c57d487d9/Os/Posix/LocklessQueue.cpp#L11 "Update candidate: LocklessQueue class") with the class template “[std::unique_ptr](https://en.wikipedia.org/wiki/Smart_pointer#unique_ptr "Description for the usage of smart pointers")”?

timcanham, 2020-02-15:
We avoid the use of any `std::` code so we don't force the user to use the standard library for small embedded processors. Adapters are free, of course, to use the standard library in their code as appropriate, but we won't put any dependencies on `std::` in the core framework code.

elfring, 2020-02-15:
How do you think about to apply a customised smart pointer then?

timcanham, 2020-02-15:
We have a flight software rule that we don't allocate memory from the heap after initialization. It can lead to non-deterministic allocation and deallocation times and leaks. If you are talking about something like a reference counting pointer that automatically deletes the memory when the references drop to zero, that risks the `free()` non-determinism. The `Fw::Buffer` and `Svc::BufferManager` pattern pre-allocates buffers from the heap at initialization and then manages them to give us a heap-like pattern.

elfring, 2020-02-16:
Which C++ class template can be used for this software to [achieve safer pointer management](https://isocpp.github.io/CppCoreGuidelines/CppCoreGuidelines#r11-avoid-calling-new-and-delete-explicitly "Guideline “R.11: Avoid calling new and delete explicitly”")?

SterlingPeet, 2020-04-01:
This framework is generally following the paradigm where memory is only allocated at startup, and deallocated at shutdown, so you shouldn't need to worry about F Prime having pointer management problems.  As @timcanham mentioned earlier, you are free to use whatever tools you want in _your_ code, but it will not get accepted into the F Prime framework upstream.

In other words, within the framework calls to ``new`` and ``delete`` are carefully avoided by during runtime design, which implements the advice in the CPP core guidelines.  Calls to ``new`` only happen during the initial object construction phase, where an error would crash the application very quickly.  Calls to ``delete`` only happen automatically as the application is shutting down, where a crash is almost irrelevant as something else has already gone wrong.

F Prime follows the [Power of 10 Rules](https://medium.com/better-programming/the-power-of-10-nasas-rules-for-coding-43ae1764f73d), and the solutions to this issue are addressed by rules  3 and 9 directly.  The entire set of rules are very useful for designing and writing reliable code, and are well worth reading and understanding.

Again, the application developer is free to use whatever tools, coding practices, and libraries they deem appropriate for their own application.  Using [std::unique_ptr](https://en.wikipedia.org/wiki/Smart_pointer#unique_ptr) in your own components is perfectly fine, if that makes your feel that your code will be better or more reliable or appropriate for you application.

elfring, 2020-04-01:
>Calls to new only happen during the initial object construction phase,

Were any update candidates still left over?

>where an error would crash the application very quickly.

Should such a view be reconsidered for the desired error response?

SterlingPeet, 2020-04-01:
> > Calls to new only happen during the initial object construction phase,

> Were any update candidates still left over?

I am not familiar with the terminology "update candidate".  However, if I assume this to mean "an object that gets destroyed and constructed from scratch during normal application runtime", then the answer is no, not really.

Objects in F Prime are generally only used to organize code that exists for the entire runtime lifecycle of the application.  This means that there are some objects that handle OS abstraction, and are not created or destroyed while running, Components, and Ports.  The Topology of components and ports is static (and global) during the runtime of the program, they get constructed before the program execution even enters the ``main()`` loop.

The major notable exception here was already explained by @timcanham, ``Fw::Buffer`` managed by ``Svc::BufferManager``.  The default implementation is carefully crafted to work in a resource constrained environment, and conforms to Power of 10 Rules _number 3_ by creating a finite, managed resource pool and making sure that if the ``Svc::BufferManager`` interface is respected no memory leaks can occur.  If a programmer chooses to create their own ``Fw::Buffer`` objects, F Prime makes no guarantees about the resulting memory usage.  This is by design, if the programmer wishes to manage the memory directly, F Prime is not stopping them.  The only guarantee is that F Prime won't cause leaks, if you use the F Prime interface.

You can trust that I understand how this works, I had to do some very in-depth debugging of the application startup process as a basis for PR #164.  And this was to support a _new_ platform (AVR ATmega), with 64 kbytes (yes, really) of system memory, and no in-system debugger capability.

>> where an error would crash the application very quickly.

> Should such a view be reconsidered for the desired error response?

Well, maybe I was unclear about this statement.  Crashing is _never_ the desired error response.  F Prime has an entire error catching system outside exception (thank goodness too, my ATmega's don't support exceptions for hardware reasons), in ``Fw::Assert``.

That said, if the only likely place for the F Prime-based software to crash is at the very beginning, then the problem is not hidden from the programmer until some random time later, after the program has been running for a long time.  The programmer is only required the due diligence of building and running the executable.

All this information pertains to the core F Prime code, not the specific things an application developer might need or want to do.  You can still use your own coding style in the custom components that you write for your project.  I am not saying that using a mechanism like [std::unique_ptr](https://en.wikipedia.org/wiki/Smart_pointer#unique_ptr) is a generally a bad idea, just that it is moot for the core F Prime code, _by design_.

elfring, 2020-04-02:
I find [the implementation of the class “LocklessQueue”](https://github.com/nasa/fprime/blob/44bc57574daaac8d455b60912f664e6c57d487d9/Os/Posix/LocklessQueue.cpp#L11 "Update candidate") questionable here.
:crystal_ball: How do you think about to improve such source code any further?

timcanham, 2020-04-02:
By all means, post any suggestions.

SterlingPeet, 2021-03-03:
This issue is stale, and can likely be closed.

LeStarch, 2021-03-04:
Agreed.  Closing for now, it can be reopened if needed.

closed 2021-03-04 by LeStarch