nasa/fprime #143: Getting Started with F' Tutorial: Cannot Exit out of Running Deployment.

sjz38, 2020-02-16:
Under the "Installing the F´ Executable and Dictionaries "Installing the F´ Executable and Dictionaries" section, after I ran the ./bin/Linux/Ref command to run the deployment, the code runs, but when I try to  CTRL+C out of the deployment, it would not exit out. Is there any way to resolve this issue?

LeStarch, 2020-02-16:
CTRL-Z
kill %1
<If still hung>
kill -KILL %1

SterlingPeet, 2020-04-01:
@LeStarch : seems like regularly needing to resort to ``kill`` is a bug.  I'll report back if I try this and see the same behavior.

LeStarch, 2020-04-01:
Usually this occurs when an active component or Task has been added without the proper call to shutdown at the end.  It bears investigation what was added to cause this.

LeStarch, 2020-05-26:
Fix should be delivered.

closed 2020-05-26 by LeStarch