nasa/fprime #144: Gettting Started with F' Tutorial: No CMakeLists for SignalGen Unit Test

sjz38, 2020-02-16:
In this tutorial, after I generate the unit test stubs for SignalGen, when I go to build the unit test of SignalGen, I recieved the following error.

make: *** No rule to make target 'Ref_SignalGen_ut_exe'. Stop.
[ERROR] CMAKE erred with return code 2.

I suspect that the issue lies in the fact that no CMakeLists document was generated. However, there is no mention of a CMakeLists in the tutorial.

LeStarch, 2020-02-16:
Currently, this problem comes from the face that the cake cache isn't
automatically rebuilt after adding CMakeLists.txt files.  To fix it, ensure
the CMakeLists.txt exists for SignalGen and is included in Ref's
CMakeLists.txt. Then run a "fprime-util purge" followed by an "fprime-util
generate" both in the deployment directory "Ref".

We'll release a fix soon that removes the need for the purge and
regeneration.  I'll also check the tutorial to make sure it requests the
addition of the CMakeLists.txt.

LeStarch, 2020-04-02:
Fix in-progress.

LeStarch, 2020-05-26:
Should be fixed.  Although we have a separate issue to clean up that tutorial, because there are not UTs for SignalGen anyway.

closed 2020-05-26 by LeStarch