nasa/fprime #145: Cheetah version error

aaronbuchwald, 2020-02-17:
When compiling, I get the error that there is an incompatible cheetah version as in the screenshot. I checked and made sure and when running 
```bash
cheetah --version
```
it says that I have cheetah version 3 installed. I haven't built anything else before, so I'm not sure why it should need to recompile with the new version. Do you know what's wrong here?
[image]

LeStarch, 2020-02-18:
This happens if you switch between Python 2 and Python 3.  ( Note, it will blow away any changes in Autocoders) Try this to fix:

```
git clean -xdf fprime/Autocoders/
```

LeStarch, 2020-02-18:
Then rebuild, and it will recompile templates.

aaronbuchwald, 2020-02-19:
Thanks, this worked!

closed 2020-02-19 by aaronbuchwald