nasa/fprime #147: Ground Station "localhost:5000" does not include GPS tutorial commands

scbrechter, 2020-02-20:
I was able to open the Ground Station html window and have it send commands to a terminal window on native MacOS(see below). However, I was not able to find the GPS tutorial commands I created in the GS drop-down list. I think I copy and pasted all your topology code from the tutorial section "Build the Topology Sources" and updated the CMakeLists.txt file correctly, but the commands do not appear in the list. Everyone on my software team who has gone through the GPS tutorial has run into the same issue.  What are we doing wrong?
[image]
[image]

LeStarch, 2020-02-20:
Thanks for posting this issue.  I am going to take a look.

LeStarch, 2020-02-20:
@scbrechter quick question: do the Channels for the GPS appear in the Channels tab above?

Double check that you are running from the "GpsApp" directory.

LeStarch, 2020-02-20:
If you have ensured you ran the "fprime-gds -d . -n" from the GpsApp directory, then make sure that the "Commands.xml' is imported in the "GpsApp/Gps/GpsComponentAi.xml" as shown below:

```
    <!-- Import command, telemetry, and event dictionaries -->
    <import_dictionary>GpsApp/Gps/Commands.xml</import_dictionary>
```

LeStarch, 2020-02-20:
Then run run the following:
```
cd GpsApp
# Clean and regenerate
fprime-util purge
fprime-util generate
# Rebuild and install
fprime-util build
fprime-util install
# Now run the ground station on the fully-rebuilt code
fprime-gds -d . -n
```

LeStarch, 2020-02-20:
Feel free to post more if you still have issues.

scbrechter, 2020-03-01:
I copied the "GpsApp/Gps/GpsComponentAi.xml" exactly from the tutorial which includes that import dictionary statement.
[image]
However I'm not seeing any GPS channels under the Channels tab
[image]
Could the issue be that I'm not actually connected to a RPI with a GPS sensor? I'm just running the ground station and other end terminal locally on my Mac. If it helps to know, Im successfully sending no_op commands to the other end terminal.

LeStarch, 2020-03-01:
If you do not have a GPS unit plugged into your mac via USB, then you should not expect channels or EVRs.  I believe the GPS component assumes that is available.

LeStarch, 2020-04-02:
Can I close this, or are there still unresolved issues?

closed 2020-05-26 by LeStarch