nasa/fprime #148: Correction to the GPS Tutorial for MacOS native users

scbrechter, 2020-02-20:
In "Running the Executable On the Native Host with the Ground System":
https://github.com/nasa/fprime/blob/master/docs/Tutorials/GpsTutorial/Tutorial.md#running-the-executable-on-the-native-host-with-the-ground-system

Please add that for MacOS users `Linux` should be replaced with `Darwin` in this command for it to run:
[image]
so MacOS users should use: 
`cd fprime/GpsApp`
`./bin/Darwin/GpsApp -a 127.0.0.1 -p 50000 -d /dev/ttyACM0`

LeStarch, 2020-02-20:
Good catch.  I'll get this fixed.

LeStarch, 2020-02-20:
@scbrechter could you look at #149.  See if the proposed clarifiactions are what you'd expect.

LeStarch, 2020-03-03:
Fixed, closing.

closed 2020-03-03 by LeStarch