nasa/fprime #150: Ground Station throws exception when leaving ground station app in browser on Mac

aaronbuchwald, 2020-02-28:
When I am running the ground station on MacOS (working off of LeStarch's gps-application branch), I run into an exception that occurs whenever I leave the ground station application. The web app starts successfully, but once I leave the web app to go back to the command line, I get the following error:

[image]

LeStarch, 2020-02-28:
That may mean the gps-application branch needs a pull from devel.  You could try that yourself, and I'll see if I can update that branch and push the updates sometime soon.

LeStarch, 2020-05-26:
I am going to move this to my fork, as that is where the update needs to occur.

LeStarch, 2020-08-03:
Needs updated GPS base app.

LeStarch, 2020-11-09:
This should be fixed.

closed 2020-11-09 by LeStarch